esp8266/Arduino #4398: Can't connect to WiFi with latest git

porkyneal, 2018-02-20:
I upgraded to the latest git to try out SDK 2.2.0 and since then the WiFi is not able to connect.

The WiFi debug output is saying:

```
reconnect
scandone"
no SSID_NAME found, reconnect after 1s
```

The SSID does exist, and was working immediately before the upgrade.
The WiFi setup routine I use:
`
WiFi.hostname(config.apSSID);   
WiFi.mode(WIFI_AP_STA); // Configure AP and Station
WiFi.softAP(config.apSSID, config.apPassword);    //set AP settings
if(config.wifiSettings.ssid != ""){  //if config has an ssid to use
     Serial.println(F("Begin WiFi"));
     WiFi.begin(config.wifiSettings.ssid, config.wifiSettings.password);
}`

The access point which is also created here works fine, once it connects (unstable while trying to connect to an access point)

I believe previously I was on core 2.4.0 and SDk 2.1.0, returning back to this everything works as expected.

Is there a anything required to clear SDK settings when switching between the two?

Using platformio if it makes any difference.

igrr, 2018-02-20:
@d-a-v and myself have noticed this as well.

Once merged, https://github.com/esp8266/Arduino/pull/4380 will offer a way to erase WiFi settings while uploading sketch.

There also needs to be a fix for this in the core, so that we can support OTA from 2.4.0 to 2.4.1.

porkyneal, 2018-02-20:
Ok, I will use 2.4.0 for now. I personally use platformio a lot. I'm guessing the menu would not work, but an erase config function will.

igrr, 2018-02-22:
Created #4412 which fixes this issue.

porkyneal, 2018-02-22:
I can confirm that this seems to have done the trick! Thanks for your quick response.

closed 2018-02-22 by porkyneal