esp8266/Arduino #4402: How to remove the handlers to reuse the server?

RazgrizX, 2018-02-20:
Have spent few hours trying to close() the server to begin () it again with new handlers but nothing works. I am adding handlers with server.on() Here is part of my code:
ESP8266WebServer server(80);
void setup() {
server.on("/def", HTTP_GET, {
server.send ( 200, "text/plain", "DefStillHere");
});
server.on("/test", HTTP_GET, {
server.send ( 200, "text/plain", "StillAlive");
});
server.begin();
}
SetMode("CONFIG_MODE");
void SetMode(String MODE) {
//works?
delay(200);
if( MODE=="CONFIG_MODE"){
//set variables
StartAP();
SetEndpoints();
}
server.begin();
}
P.S. just want to remove the handlers, so I wont have to check if they should be available.

devyte, 2018-02-20:
@RazgrizX the handlers are set in the webserver object. Calling close() closes the server, it doesn't reset the webserver object, which means the handlers are still there.
Calling begin() restarts the server, and then the set handlers come into play.
Closing as intended behavior.

closed 2018-02-20 by devyte

RazgrizX, 2018-02-21:
Thanks, I get it. But isn't there a way to reset the server or recreate it with the same name?