esp8266/Arduino #4405: Getting Exception (3) when trying to send data

alexrv42, 2018-02-21:
### Basic Infos

#### Hardware
Hardware:			ESP8266
Core Version:      	1.0.0

### Description

I am getting Exception (3) when I try to send data to a AWS server. I have made sure that my credentials are ok so I have to ask you for help. Below is my stack trace. 

I'm using the following library: https://github.com/heskew/aws-sdk-arduino/commits/master 

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		512k
CPU Frequency:        80Mhz
Flash Mode:		qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch

Sketch with dummy credentials:

```
#include <ESP8266WiFi.h>
#include <ESP8266AWSImplementations.h>
#include <AmazonIOTClient.h>

Esp8266HttpClient httpClient;
Esp8266DateTimeProvider dateTimeProvider;

AmazonIOTClient iotClient;
ActionError actionError;

const char* ssid = "ssid";
const char* password = "password";

void initWLAN()
{
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
  }
}

void initAWS()
{
  iotClient.setAWSRegion("us-west-2");
  iotClient.setAWSEndpoint("amazonaws.com");
  iotClient.setAWSDomain("az3y9qqx4lf51.iot.us-west-2.amazonaws.com");
  iotClient.setAWSPath("/things/ESP8266/shadow");
  iotClient.setAWSKeyID("my key");
  iotClient.setAWSSecretKey("secret key");
  iotClient.setHttpClient(&httpClient);
  iotClient.setDateTimeProvider(&dateTimeProvider);
}

void setup() {
  Serial.begin(115200);
  delay(10);
  Serial.println("begin");
  initWLAN();
  Serial.println("wlan initialized");
  initAWS();
  Serial.println("iot initialized");
}

void loop()
{
  char shadow[100];
  strcpy(shadow, "{\"state\":{\"reported\":{\"test_value1\":123, \"test_value2\":234}}}");
  
  Serial.println("Trying to send data");
  Serial.print(shadow);
  
  char* result = iotClient.update_shadow(shadow, actionError);
  Serial.print(result);
  
  delay(2000);
}

```

### Debug Messages

This is my stack trace in the EspDecoder

```

0x4010020c: _umm_free at C:\Users\alexr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266\umm_malloc/umm_malloc.c line 1291
0x40100690: free at C:\Users\alexr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266\umm_malloc/umm_malloc.c line 1737
0x40206490: operator delete[](void*) at C:\Users\alexr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/abi.cpp line 84
0x40204095: MinimalString::innerDelete() at C:\Users\alexr\Documents\Arduino\libraries\common/AWSFoundationalTypes.cpp line 49
0x402040f9: MinimalString::operator=(MinimalString const&) at C:\Users\alexr\Documents\Arduino\libraries\common/AWSFoundationalTypes.cpp line 49
0x4020647c: operator delete(void*) at C:\Users\alexr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/abi.cpp line 84
0x40203f50: AWSClient4::createRequest(MinimalString&) at C:\Users\alexr\Documents\Arduino\libraries\common/AWSClient4.cpp line 197
0x4021fe0d: SHA256_Final at crypto/sha256.c line 273 (discriminator 3)
0x4021c26c: hmac_sha256_v at crypto/hmac.c line 165
0x4021c2b8: ax_hmac_sha256 at crypto/hmac.c line 133
0x40216165: p_hash_sha256 at ssl/tls1.c line 2007
:  (inlined by) prf at ssl/tls1.c line 951
0x40103215: lmacProcessTXStartData at ?? line ?
0x40103212: lmacProcessTXStartData at ?? line ?
0x4010213e: wDev_ProcessFiq at ?? line ?
0x40100ece: pp_post at ?? line ?
0x401042b0: lmacTxFrame at ?? line ?
0x401034cf: lmacRecycleMPDU at ?? line ?
0x40103932: lmacRecycleMPDU at ?? line ?
0x40103416: lmacProcessTxSuccess at ?? line ?
0x40102297: wDev_ProcessFiq at ?? line ?
0x40101f94: wDev_ProcessFiq at ?? line ?
0x4021a4ba: pgm_read_byte at ssl/os_port.h line 107
:  (inlined by) AES_encrypt at crypto/aes.c line 375
:  (inlined by) AES_cbc_encrypt at crypto/aes.c line 288
0x4021a3be: AES_cbc_encrypt at crypto/aes.c line 283
0x4024867e: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x402485d0: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x4021ae5a: bi_free at crypto/bigint.c line 672
0x4021adf8: bi_depermanent at crypto/bigint.c line 672
0x4010020c: _umm_free at C:\Users\alexr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266\umm_malloc/umm_malloc.c line 1291
0x4010020c: _umm_free at C:\Users\alexr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266\umm_malloc/umm_malloc.c line 1291
0x40100690: free at C:\Users\alexr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266\umm_malloc/umm_malloc.c line 1737
0x4021af18: bi_terminate at crypto/bigint.c line 672
0x4010020c: _umm_free at C:\Users\alexr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266\umm_malloc/umm_malloc.c line 1291
0x4010020c: _umm_free at C:\Users\alexr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266\umm_malloc/umm_malloc.c line 1291
0x4010020c: _umm_free at C:\Users\alexr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266\umm_malloc/umm_malloc.c line 1291
0x40100690: free at C:\Users\alexr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266\umm_malloc/umm_malloc.c line 1737
0x4020647c: operator delete(void*) at C:\Users\alexr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/abi.cpp line 84
0x40203668: SSLContext::unref() at C:\Users\alexr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 110
0x40203692: WiFiClientSecure::~WiFiClientSecure() at C:\Users\alexr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 415
0x402044a9: Esp8266HttpClient::send(char const*, char const*, int) at C:\Users\alexr\Documents\Arduino\libraries\common/ESP8266AWSImplentations.cpp line 54
0x402069a8: Print::write(unsigned char const*, unsigned int) at C:\Users\alexr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/Print.cpp line 38
0x401004d8: malloc at C:\Users\alexr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266\umm_malloc/umm_malloc.c line 1668
0x40206468: operator new[](unsigned int) at C:\Users\alexr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/abi.cpp line 84
0x40204199: AmazonIOTClient::update_shadow(MinimalString, ActionError&) at C:\Users\alexr\Documents\Arduino\libraries\common/AmazonIOTClient.cpp line 19
0x4020226a: loop at C:\Users\alexr\Documents\Arduino\ESP8266AWSIoTDemo-master\ESP8266AWSIoTDemo/ESP8266AWSIoTDemo.ino line 74
0x402025f8: ESP8266WiFiSTAClass::status() at C:\Users\alexr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.cpp line 483
0x40205e1c: Print::println() at C:\Users\alexr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/Print.h line 93
0x40205e40: Print::println(char const*) at C:\Users\alexr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/Print.h line 93
0x40206565: esp_schedule at C:\Users\alexr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/core_esp8266_main.cpp line 57
0x40206590: loop_wrapper at C:\Users\alexr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/core_esp8266_main.cpp line 57
0x40100718: cont_norm at C:\Users\alexr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/cont.S line 109

```

d-a-v, 2018-02-21:
We can't debug third-party libraries here.
Please activate all debug options in Tools menu, including OOM, and print out you console results along with your stack trace so you can check whether your issue is core-related.

devyte, 2018-02-21:
Core version 1.0.0? Maybe you want to try a later one...

Anton-V-K, 2018-02-25:
@devyte , the stack trace shows that actually 2.4.0 is used, so `1.0.0` is probably a typo...

devyte, 2018-02-25:
@alexrv42 please retry with latest git. There have been several critical fixes merged since release 2.4.0.