esp8266/Arduino #4407: PinMode(s) INPUT/OUTPUT

Tensa225, 2018-02-21:
Switching the PinMode(s) within a sketch does not seem to function properly. When moving back to PinMode OUTPUT from INPUT it interrupts or changes to GND level every second or so (when using analogwrite for PWM).

When using OUTPUT mode only there does not seem to be a problem except that this mode is high impedance.

Switching to INPUT mode can be a requirement for sinking current. The high impedance state when using OUTPUT mode can cause oscillations when using level shifters for instance. As the IO can`t correctly pull the external device to GND level.

On most boards OUTPUT mode should prevent the IO to go into a low impedance state as the current can possibly damage the device.

Please confirm INPUT is actually low impedance, OUTPUT is high impedance and that switching the PinMode inside the sketch is possible.

I did not test the OUTPUT pull-up mode.

Tested version is 2.4.0 (previous versions seemed to behave differently)

WereCatf, 2018-02-21:
Why do you make a new thread about this? You were already told previously that INPUT is high-impedance, OUTPUT is low-impedance.

devyte, 2018-02-21:
Closing as duplicate of #4246 . Interesting that the original post there was edited and removed 6 days ago.

closed 2018-02-21 by devyte

devyte, 2018-02-21:
An article that may help understand pin modes in a general sense (not specific to the ESP):
https://stratifylabs.co/embedded%20design%20tips/2013/10/21/Tips-Understanding-Microcontroller-Pin-Input-Output-Modes/

I suspect @Tensa225 says "OUTPUT high impedance" but really means open drain mode.
In any case, discussion of this should be done at a forum, and not here.

vdeconinck, 2018-02-21:
@Tensa225 you must be kidding. Did you bother reading answers in #4246 ?

Tensa225, 2018-02-22:
The answers are all incorrect and it was closed for no reason.

They propably wrote that article to fit their ESP theology.