esp8266/Arduino #4410: ESP ignores its internal DNS table to resolve hosts and always makes a new query

ranr, 2018-02-21:
I've noticed that the ESP8266's version of LWIP has line 977 in dns.c commented out:
//    ipaddr = dns_lookup(hostname);

This means that the ESP8266 effectively ignores the internal DNS table and will always make a DNS query. I couldn't trace the commit or understand why this wasn't implemented.

In many cases the DNS may fall from time to time and making an active DNS call on every query makes the ESP8266 less robust and more prone to temporary DNS issues.

Anyone knows why this is so? @me-no-dev @igrr 

Thanks!

d-a-v, 2018-02-21:
@ranr good finding in lwip-v1.4.
Default current version is lwIP-v2 and the code you are referring to is [there](http://git.savannah.nongnu.org/cgit/lwip.git/tree/src/core/dns.c?h=STABLE-2_0_3_RELEASE&id=92f23d6ca0971a32f2085b9480e738d34174417b#n1531).
You are however still right because the dns cache size is [rather small](https://github.com/d-a-v/esp82xx-nonos-linklayer/blob/master/glue-lwip/arduino/lwipopts.h#L1043).
You are free to rebuild lwip2 with updated parameters with in mind that static (dns) tables will steal your ram. Note that nobody here is currently maintaining lwIP-v1.4. The SDK repo would be the place to ask.

d-a-v, 2018-02-22:
closing as [intentional](https://github.com/espressif/ESP8266_NONOS_SDK/blob/fcdedd636cc847d54f5576e804e4cde7e1147ea5/third_party/lwip/core/dns.c#L977) and unsupported.

closed 2018-02-22 by d-a-v