esp8266/Arduino #4411: ESP8266 panic on v2.4.0

mars000, 2018-02-22:
Hi

I am experiencing below PANIC errors. I can't seem to work out why.
I can confirm that when I roll back to Core V2.4.0RC2 it all works fine. This PANIC error only seems to occur in V2.4.0

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			ESP8266EX - wemos mini pro
Core Version:      	2.4.0

### Description

Problem description

### Settings in IDE

Module:  			wemos mini pro
Flash Size: 		16MB
CPU Frequency:		*0Mhz
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch
BELOW IS CALLING CODE....function where error occurring is within readResponseContent function which is also shown below.

```cpp

firmware_check check_firmware_version(char *json_server, char *json_resource, int local_CURRENT_VERSION)
{
	WiFiClient client_json;
	firmware_check return_result;
	return_result.error_detected = false;
	if (connect(json_server, &client_json))
	{ // if1
		if (sendRequest(json_server, json_resource, &client_json) && skipResponseHeaders(&client_json))
		{ // if2
			json_file_data jsondata;

			if (readResponseContent(&jsondata, &client_json)) // was &client_json
			{ // if3

				Serial.print(F("Current Firmware Version = "));
```
```cpp

bool readResponseContent(struct json_file_data *jsondata, WiFiClient *client_json) // was *client_json
{
	// Compute optimal size of the JSON buffer according to what we need to parse.
	// This is only required if you use StaticJsonBuffer.

	DynamicJsonBuffer jsonBuffer;

#ifdef debug_json
	Serial.println(F("waiting for parsing to occur..."));
	blynk_status (262, "waiting for parsing to occur", true);
#endif

	JsonObject &root = jsonBuffer.parseObject(*client_json);  // TIIS IS WHERE PANIC ERROR OCCURS

```

### Debug Messages
SERIAL OUTPUT BELOW:
```
I'm just before client_json.available()
waiting for parsing to occur...
 line : 262 waiting for parsing to occur
Heap size = 22944

Panic C:\Users\mdavo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266\core_esp8266_main.cpp:133 loop_task

ctx: sys 
sp: 3ffffdd0 end: 3fffffb0 offset: 01b0

>>>stack>>>
3fffff80:  3fff657c 3fff656c 00000000 3fff37a0  
3fffff90:  00000000 3fffdad0 3fff3790 40219399  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```
messages here
```

mars000, 2018-02-22:
hi - the sketch is huge. I'm not sure how I will extract. I will try.
I am using the very latest git which I downloaded 2 hrs ago.

mars000, 2018-02-22:
i've done a bit more work.
I can confirm I get a PANIC error when I use latest V2.4.0 with LwIP Variant V 2.0 Prebuilt mSS=536 
However it all works fine when I use V1.4 Prebuild. Is this a known issue ?
See error below:
[image]

mars000, 2018-02-22:
sorry to place another post. I collected a more descriptive output, the problem is I don't know how to decode it. Any clues to the issue and how I would best educate myself on how to read this output ?

```

```
[image]
[image]

devyte, 2018-02-22:
Your stack dump shows use of release 2.4.0. Like I said, please try *latest git*.

devyte, 2018-03-06:
@mars000 did you retest with latest git? as in latest state of the master branch, and not the last release.

devyte, 2018-07-13:
Closing due to lack of feedback in several months.

closed 2018-07-13 by devyte