esp8266/Arduino #4414: ESP8266 Error: warning: espcomm_sync failed error: espcomm_open failed error: espcomm_upload_mem failed

HarukiHirosan, 2018-02-22:
I am using an Ardunio Genuino 101 Board to upload the sketch into my ESP module. Without any USB-TTL converter. I am not using any voltage regulators. My layout is as following on the image below but without the voltage regulator. Just straight connection to 3.3v port on the ardunio board. I've tried using an external 3.3v power but it didn't work either.
[image]

Hardware:	ESP8266-01

Whenever I try to upload a sketch, it gives me this error:

warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

I do not want to buy an USB-TTL converter just for this. I can't upload anything on the module.

Module:  			Generic ESP8266 Module
Flash Size: 		1M(512k SPIFFS)
CPU Frequency:	80Mhz
Flash Mode:		QIO
Flash Frequency:	40Mhz
Upload Using:		USB cable via Arduino Genuino 101
Reset Method:		ck

Sketch:

/*
 *  This sketch demonstrates how to scan WiFi networks. 
 *  The API is almost the same as with the WiFi Shield library, 
 *  the most obvious difference being the different file you need to include:
 */
#include "ESP8266WiFi.h"

void setup() {
  Serial.begin(115200);

  // Set WiFi to station mode and disconnect from an AP if it was previously connected
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  delay(100);

  Serial.println("Setup done");
}

void loop() {
  Serial.println("scan start");

  // WiFi.scanNetworks will return the number of networks found
  int n = WiFi.scanNetworks();
  Serial.println("scan done");
  if (n == 0)
    Serial.println("no networks found");
  else
  {
    Serial.print(n);
    Serial.println(" networks found");
    for (int i = 0; i < n; ++i)
    {
      // Print SSID and RSSI for each network found
      Serial.print(i + 1);
      Serial.print(": ");
      Serial.print(WiFi.SSID(i));
      Serial.print(" (");
      Serial.print(WiFi.RSSI(i));
      Serial.print(")");
      Serial.println((WiFi.encryptionType(i) == ENC_TYPE_NONE)?" ":"*");
      delay(10);
    }
  }
  Serial.println("");

  // Wait a bit before scanning again
  delay(5000);
}

Please help me.

WereCatf, 2018-02-22:
You need to boot the ESP into flash-mode before you can upload any sketches onto it, ie. push flash-button down, press once on the reset-button, then let go of the flash-button.

HarukiHirosan, 2018-02-22:
I already tried that and there is no diffrence to it. Perhaps something is wrong? When I do the flash mode thing the blue led flashes once. The red led is always on...

devyte, 2018-02-23:
Off topic, please request assistance at a community forum like esp8266.com or stackoverflow.
Closing per issue POLICY doc.
Btw, you need a special sketch on the UNO to use it like that.

closed 2018-02-23 by devyte

arghasen10, 2018-05-27:
I was also facing the same problem. The solution is you need to connect the reset pin of your Arduino UNO to ground and the GPIO0 of esp8266 to a button which is already connected to ground. Now before uploading the code in esp8266, you need to remove the USB and then keep the button pressed so that the GPIO0 is grounded and then connect the USB and then just unpress it. Next time when you are going to upload your code again you have to follow the same steps.

ravindrasitale096, 2018-07-31:
Thank You arghasen10, it really worthy information.

darrahts, 2019-02-07:
I know the issue is closed but there should be no reason to go through all of the stuff mentioned here. First time using the nodeMCU's I got the blinky sketch to work, and then something happened between my laptop updating and developing on different machines. I spent days trying to figure out why some of my old sketches, when edited or deployed to new NodeMCUs would work, and nothing else would.
```
**Board: "NodeMCU 1.0 (ESP-12E Module)"
Upload Speed: "115200"
CPU Frequency: "80 MHz"
Flash Size: "4M (no SPIFFS)"
Debug port: "Disabled"
Debug Level: "None"
lwIP Variant: "v2 Lower Memory"
VTables: "Flash"
Exceptions: "disabled"
Erase Flash: "Only Sketch"
Port: <your_port>**
```

raymop, 2019-02-09:
> I know the issue is closed but **there should be no reason to go through all of the stuff mentioned here**. First time using the nodeMCU's I got the blinky sketch to work, and then something happened between my laptop updating and developing on different machines. I spent days trying to figure out why some of my old sketches, when edited or deployed to new NodeMCUs would work, and nothing else would.
> 
> **The following configuration solved the issue:**
> 
> ```
> Board: "NodeMCU 1.0 (ESP-12E Module)"
> Upload Speed: "115200"
> CPU Frequency: "80 MHz"
> Flash Size: "4M (no SPIFFS)"
> Debug port: "Disabled"
> Debug Level: "None"
> lwIP Variant: "v2 Lower Memory"
> VTables: "Flash"
> Exceptions: "disabled"
> Erase Flash: "Only Sketch"
> Port: <your_port>
> ```
> This worked across different machines, using[ this link](http://arduino.esp8266.com/stable/package_esp8266com_index.json) to add the board file to boards manager. I downloaded "**esp8266** by **ESP8266 Community** version 2.5.0".

I have tried what you showed above, and really, really hoped it would work.
I have several HUZZAH feathers. Including one that I uploaded firmware to last year.
The firmware is still running, and When I view the serial monitor, I get the output expected.
But, I cannot, cannot upload any new firmware to that HUZZAH or any new ones I bought.
Have tried everything,
- different PCs (Mac vs WIN10)
- different serial cables
- speeds
- sketches
- Arduino IDE vs pio

And I always get the same error.:
Error: warning: espcomm_sync failed error: espcomm_open failed error: espcomm_upload_mem failed

I'm about to throw away four ESP8266's and try ESP-32.

What am I missing????!

darrahts, 2019-02-09:
@raymop the feather requires additional drivers, have you followed the guide here https://learn.adafruit.com/adafruit-feather-huzzah-esp8266/using-arduino-ide ?

d-a-v, 2019-02-09:
The official flasher is esptool.py

https://github.com/espressif/esptool

We will use it after #5635 is merged.

Try it, or try #5635.

eilering, 2022-12-26:
I tried everything. I went through all the stuff in https://arduino-esp8266.readthedocs.io/en/latest/faq/a01-espcomm_sync-failed.html. No result. The only thing left was a scope. I tried 5 ESP modules, 4 Arduino Uno. Still this failure. IT HAD TO BE A SOFTWARE PROBLEM
I removed the entire installation of Arduino with revo uninstaller. It looked as if there were two installations, an Arduino and an Arduino IDE. and did a new installation of Arduino IDE 2.0.3. During this new Arduino installation it also installed an ESP board. Strange. How did he know. The installed ESP board was version 3.0.2. Still no result. Then I installed version 2.5.0 of the esp board. IT WORKS.

Bighoneypot, 2022-12-26:
> I am using an Ardunio Genuino 101 Board to upload the sketch into my ESP module. Without any USB-TTL converter. I am not using any voltage regulators. My layout is as following on the image below [...]

Add #include "Arduino.h"
and Add delay(150) after WiFi.mode(WIFI_STA);

eilering, 2022-12-29:
did you try my experience? See previous note.
_When i commented this lline, the upload to the esp worked. ( //Serial.begin(115200); )
when I uncommented it, ( Serial.begin(115200); ) the upload to the esp failed._

Bighoneypot, 2022-12-29:
Add #define Arduino.h and check your cable. You need to check your upload methods.