esp8266/Arduino #4415: Invalid wifi hostname

leifnel, 2018-02-22:
The hostname sent in the dhcp request contains an underscore; ESP_123456

This causes DDNS updates to fail, as underscore is not legal in hostname (except as first character)

 dhcpd[32702]: Unable to add forward map from ESP_123456.lan. to 192.168.1.152: REFUSED

Solution: Just omit the underscore.

leifnel, 2018-02-23:
This is a workaround:

    ` String hostname = WiFi.hostname();
      hostname[3]='-';
      WiFi.hostname(hostname);
      WiFi.begin(ssid, password);`

devyte, 2018-02-27:
@leifnel when you opened the issue, you were presented with a template requesting specific info. You ignored it its entirety. Please edit your post and fill it in.
Are you talking about an example sketch, your own app, or something that is hardcoded in the libs?

devyte, 2018-03-02:
@leifnel Will you be providing the info requested in the issue template? At the very least an example sketch is needed.

leifnel, 2018-03-02:
The relevant code is simply

```
#include <ESP8266WiFi.h>
void setup() {
  WiFi.mode(WIFI_STA);
  WiFi.begin("ssid", "password");
}
```
The hostname is in the format ESP_123456
When the device is connecting to a wifi, where the dhcp-server tries to update the bind9 dns-server with the hostname, the ESP_123456 name is denied, because underscore is not a valid character in hostnames.

igrr, 2018-03-02:
This looks like an SDK bug, but we can work around it somewhere in WiFi.begin.

Makuna, 2018-10-05:
A better place would be in ESP8266WiFiGenericClass::enableSTA() as it is common to the at least paths to configure and setup WiFi.

d-a-v, 2018-12-30:
This recently came to my attention (I had searched for this before, it was too hidden):
```
1. A "name" (Net, Host, Gateway, or Domain name) is a text string up
   to 24 characters drawn from the alphabet (A-Z), digits (0-9), minus
   sign (-), and period (.).  Note that periods are only allowed when
   they serve to delimit components of "domain style names". (See
   RFC-921, "Domain Name System Implementation Schedule", for
   background).  No blank or space characters are permitted as part of a
   name. No distinction is made between upper and lower case.  The first
   character must be an alpha character.  The last character must not be
   a minus sign or period.  A host which serves as a GATEWAY should have
   "-GATEWAY" or "-GW" as part of its name.  Hosts which do not serve as
   Internet gateways should not use "-GATEWAY" and "-GW" as part of
   their names. A host which is a TAC should have "-TAC" as the last
   part of its host name, if it is a DoD host.  Single character names
   or nicknames are not allowed.
```
RFCs:
[Internet host table specification](https://www.ietf.org/rfc/rfc952.txt) (1985)
ref-ed from [ Requirements for Internet Hosts -- Application and Support](https://www.ietf.org/rfc/rfc1123.txt) (1989, section 2.1 Host Names and Numbers)

Hostnames are set in lwIP interfaces (`netif`, one for STA, one for AP, at least).
They become visible in the network on a DHCP request (STA as a client). That's why hostname must be set up very early in the sketch (a DHCP request is silently done shortly after reset in persistence mode).

For that reason, a fixing check should be done in lwip2 (when [setting up network interfaces](https://github.com/d-a-v/esp82xx-nonos-linklayer/blob/56e04100107366601e6fbf2bea55c32b390a98c3/glue-lwip/lwip-git.c#L312)), and also in [`WiFi.hostname(name)`](https://github.com/esp8266/Arduino/blob/b26c19e82e1b6c94f86a66dc2652336e8ca5bba3/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp#L479) by a common function in the core that would replace offending chars / fix hostname validity.

leifnel, 2018-12-30:
An (unrelated) comment on this: some nameservers (or GUI's) incorrectly prevent entries, which are not hostnames, to have names like _sip.example.org.

devyte, 2019-02-05:
Reference:
https://github.com/espressif/ESP8266_NONOS_SDK/issues/208

PR #5873 (2019-03-14): update firmware to espressif branch nonos-sdk 2.2.x (19.03.13)
update firmware to espressif branch nonos-sdk 2.2.x (19.03.13)

fix #4415 

on NONOS-SDK branch release-2.2.x: `git log 509eae8..c7b580c`

*edit*: please note that I did not filter those logs, lots of things we don't use (AT, ssl, ...)

commit c7b580c679e45d6c0a8f41c9b0913837b76ad4db
Merge: 183d43b 476e1a1
Author: Xu Chun Guang <xuchunguang@espressif.com>
Date:   Wed Mar 13 11:32:22 2019 +0800

    Merge branch 'feature/remove_phy_freq_trace' into 'release/v2.2.x'

    feat: Remove phy freq trace api and disable freq trace in at pr [... cut]
closed 2019-03-14 by d-a-v