esp8266/Arduino #4416: DNS setting for SoftAP

chrzimmer, 2018-02-22:
Hello, 

is there any possibility to set the address of a dns server ? I expected that i can set this in WiFI.softAPConfig like the APs ip, the subnet and the gateways IP.

The AP always sets the DHCP and DNS address to itself. But this is only correct for DHCP.

This is really an urgent request since i am late in my project :(

THX in advance

chrzimmer, 2018-02-22:
can i make it deep down in the code the same address like the gateway ? This would also help

devyte, 2018-02-27:
I'd say that you're using a rather special network setup if your dns server is somehow supposed to be found through a client that is connected to the softAP. 
From looking at the ::softAPConfig() code, and user_interface.h SDK file, I don't see a straightforward way of implementing this (i.e.: it's not a matter of just exposing some SDK api). 
If your need is to use as dns server the one specified by the router that the STA side is connected to, then currently the only straightforward way I know to do that is to set up a DNS forwarder on the ESP.
Having said the above, it should be possible to set up the DNS server address with the DHCP options that are set up inside ::softAPConfig(). However, I consider a change like this, where code complexity is increased only to cover a special network setup that few would use, to be very low priority, and hence it is unlikely that this issue will be addressed in the near future.
I'm closing this as won't fix for now. If you wish to investigate this yourself, I suggest installing latest git and playing with the code mentioned above. If you do find an answer, feel free to contribute it with a PR, so that it can be reviewed.

closed 2018-02-27 by devyte