esp8266/Arduino #4423: Problem with ESP8266 Get Request and DMD2 Library

viceplayer, 2018-02-23:
Hello Dear All,

I am using ESP8266 to download information from server using simple get Request and DMD2 library to draw them on display.

Problem I have is, that when I call dmd.begin() after some time ESP8266 will become unable to connect to wifi, sometimes it even causes watchdog resets wdt.

I found out that both of them use Interrupts. It might be a problem, I think.

Could you help me to fix it?
[code.zip](https://github.com/esp8266/Arduino/files/1752138/code.zip)

devyte, 2018-02-27:
@viceplayer When you opened the issue you were presented with an issue template requesting specific info. You ignored it in its entirety.
Is this a duplicate of #4244 ? Also, In case you're working with 2.4.0, I suggest retrying with latest git, as there have been several fixes merged after release 2.4.0 which could be relevant.

devyte, 2018-03-02:
Closing as duplicate of #4244 .

closed 2018-03-02 by devyte