esp8266/Arduino #4424: POST to ESP8266HTTPUpdateServer requires authentication

f0nt4, 2018-02-23:
I've created a custom submit form for uploading a .bin file.
But when I POST to /update the page requests authentication, whatever I fill in, it won't accept it.
Also using the library without username and password still requires to authenticate.

This is how I setup the library:
```cpp
ESP8266WebServer webserver(80);
ESP8266HTTPUpdateServer httpUpdater;

const char* update_path = "/update";
const char* update_username = "admin";
const char* update_password = "admin";

void setup_webserver() {
  
  //setup webserver addresses
  webserver.on("/", web_returnRoot);
  webserver.on("/upload", web_updatePage);

  httpUpdater.setup(&webserver, update_path, update_username, update_password);
  webserver.begin();
}
```

And the page with the form is as follows:
```cpp
void web_updatePage() {
  String message = "<html lang='en'><head><meta charset='utf-8'><meta name='viewport' content='width=device-width, initial-scale=1, shrink-to-fit=no'><title>Doorbell Configuration</title>";
  message += "<link rel='stylesheet' href='https://maxcdn.bootstrapcdn.com/bootstrap/4.0.0/css/bootstrap.min.css' integrity='sha384-Gn5384xqQ1aoWXA+058RXPxPg6fy4IWvTNh0E263XmFcJlSAwiGgFAW/dAiS6JXm' crossorigin='anonymous'>";
  message += "</head><body>";
  message += "<nav class='navbar navbar-expand-md navbar-dark bg-dark mb-4'><div class='container'><a href='/' class='navbar-brand'>Doorbell Configuration</a></div></nav>";
  message += "<main role='main' class='container'><div class='jumbotron'>";
  message += "<form id='data' action='/update' method='POST'>";
  message += "<div class='input-group mb-3'><div class='custom-file'><input type='file' accept='.bin' class='custom-file-input' id='inputGroupFile04' name='update'><label class='custom-file-label' for='inputGroupFile04'>Choose file</label></div><div class='input-group-append'><button class='btn btn-danger' type='submit'>Update!</button></div></div>";
  message += "<div class='form-group'><button type='button' class='btn btn-Primary' onclick='window.location.href=\"/\"'>Back</button></div>";
  message += "</form></div></main></body></html>";
  
  webserver.send(200, "text/html", message);
}
```
I've also tried to put authentication on the /upload page and authenticate with the username and password and then submitting a file. But it will still request authentication.

Is there anything I can do to make this form work?
[image]

f0nt4, 2018-02-23:
I've created a PR: https://github.com/esp8266/Arduino/pull/4425
It might not be the best solution, but it does fix not being able to authenticate when I use a custom form.

f0nt4, 2019-05-08:
Nobody cares. Closing the issue.

closed 2019-05-08 by f0nt4

arlaor, 2020-01-10:
Hi @Fonta 

I don't know if the issue applies to this problem, but I will ask. I have uploaded three pages html through FS.h to the memory of esp8266, among which is the page that updates the code of esp8266, but when calling the file .bin, it does not update the code of esp8266. What extra code should I write to get the update information of the page mounted in the ESP memory?

This is my code

```javascript
#include <ESP8266HTTPUpdateServer.h>
#include <ESP8266WiFiMulti.h>
#include <WebSocketsServer.h>
#include <ESP8266mDNS.h>
#include <Hash.h>

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>

#define DEBUG_SERIAL //uncomment for Serial debugging statements
#ifdef DEBUG_SERIAL
#define DEBUG_BEGIN Serial.begin(115200)
#define DEBUG_PRINT(x) Serial.println(x)
#else
#define DEBUG_PRINT(x)
#define DEBUG_BEGIN
#endif
//includes
#include <PersWiFiManager.h>
#include <ArduinoJson.h>
#include <EasySSDP.h> // http://ryandowning.net/EasySSDP/
//extension of ESP8266WebServer with SPIFFS handlers built in
#include <SPIFFSReadServer.h> // http://ryandowning.net/SPIFFSReadServer/
// upload data folder to chip with Arduino ESP8266 filesystem uploader
// https://github.com/esp8266/arduino-esp8266fs-plugin
#include <DNSServer.h>
#include <FS.h>
#define DEVICE_NAME "ESP8266 DEVICE"
//server objects
SPIFFSReadServer server(80);
DNSServer dnsServer;
PersWiFiManager persWM(server, dnsServer);

const char* host = "Light";
const char* update_path = "/update";
const char* update_username = "xx";
const char* update_password = "xxxxxxx";

IPAddress ip = WiFi.localIP(); // desired IP Address
IPAddress mask = WiFi.subnetMask(); //Mascara de red
IPAddress gat = WiFi.gatewayIP();//gateway

MDNSResponder mdns;
ESP8266WiFiMulti WiFiMulti;
ESP8266WebServer httpServer(80);
ESP8266HTTPUpdateServer httpUpdater;

Fonta, 2020-01-13:
Hi @arlaor 

My quick guess would be that `httpServer.begin();` has to be after `httpUpdater.setup(&httpServer, update_path, update_username, update_password);`.

Otherwise I don't really see anything.