esp8266/Arduino #4427: Memory drain if webserver not served

s0170071, 2018-02-24:
Using the ESP8266 WebServer I found that if I do not call webserver.handleClient() often enough, incoming packets start to drain the RAM. While thats ok until a certain point, it should,  not crash the ESP. However, it does. 

Is there a web server parameter that’s keeping the ESP from storing incoming packets if memory is low ?

devyte, 2018-02-24:
What does not calling it enough mean exactly? I don't understand your issue, please explain in more detail and with an example.
The handleClient method is the dispatcher that (in part) allows accepting data that has been received by lwip, which needs to be handed off from lwipn to the application asap. If you don't call it frequently enough, the lwip buffers blow up after accumulating too much.
Also, if you're doing frequent http requests, you should either reuse the http connection, or clean up with tcp_abort (TIME_WAIT).

s0170071, 2018-02-24:
Here is an example. Its the advanced Webserver example with a modified loop()
If you send requests to it , memory goes way down. (LWIP 2) 
LWIP1 seems ok, memory drops at first but comes back eventually.

```
 

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>

const char *ssid = "";
const char *password = "";

ESP8266WebServer server ( 80 );

const int led = 13;

void handleRoot() {
	digitalWrite ( led, 1 );
	char temp[400];
	int sec = millis() / 1000;
	int min = sec / 60;
	int hr = min / 60;

	snprintf ( temp, 400,

"<html>\
  <head>\
    <meta http-equiv='refresh' content='5'/>\
    <title>ESP8266 Demo</title>\
    <style>\
      body { background-color: #cccccc; font-family: Arial, Helvetica, Sans-Serif; Color: #000088; }\
    </style>\
  </head>\
  <body>\
    <h1>Hello from ESP8266!</h1>\
    <p>Uptime: %02d:%02d:%02d</p>\
    <img src=\"/test.svg\" />\
  </body>\
</html>",

		hr, min % 60, sec % 60
	);
	server.send ( 200, "text/html", temp );
	digitalWrite ( led, 0 );
}

void handleNotFound() {
	digitalWrite ( led, 1 );
	String message = "File Not Found\n\n";
	message += "URI: ";
	message += server.uri();
	message += "\nMethod: ";
	message += ( server.method() == HTTP_GET ) ? "GET" : "POST";
	message += "\nArguments: ";
	message += server.args();
	message += "\n";

	for ( uint8_t i = 0; i < server.args(); i++ ) {
		message += " " + server.argName ( i ) + ": " + server.arg ( i ) + "\n";
	}

	server.send ( 404, "text/plain", message );
	digitalWrite ( led, 0 );
}

void setup ( void ) {
	pinMode ( led, OUTPUT );
	digitalWrite ( led, 0 );
	Serial.begin ( 115200 );
	WiFi.mode ( WIFI_STA );
	WiFi.begin ( ssid, password );
	Serial.println ( "" );

	// Wait for connection
	while ( WiFi.status() != WL_CONNECTED ) {
		delay ( 500 );
		Serial.print ( "." );
	}

	Serial.println ( "" );
	Serial.print ( "Connected to " );
	Serial.println ( ssid );
	Serial.print ( "IP address: " );
	Serial.println ( WiFi.localIP() );

	if ( MDNS.begin ( "esp8266" ) ) {
		Serial.println ( "MDNS responder started" );
	}

	server.on ( "/", handleRoot );
	server.on ( "/test.svg", drawGraph );
	server.on ( "/inline", []() {
		server.send ( 200, "text/plain", "this works as well" );
	} );
	server.onNotFound ( handleNotFound );
	server.begin();
	Serial.println ( "HTTP server started" );
}

void loop ( void ) {
	//server.handleClient();
  delay(1000); 
  Serial.println(ESP.getFreeHeap());
}

void drawGraph() {
	String out = "";
	char temp[100];
	out += "<svg xmlns=\"http://www.w3.org/2000/svg\" version=\"1.1\" width=\"400\" height=\"150\">\n";
 	out += "<rect width=\"400\" height=\"150\" fill=\"rgb(250, 230, 210)\" stroke-width=\"1\" stroke=\"rgb(0, 0, 0)\" />\n";
 	out += "<g stroke=\"black\">\n";
 	int y = rand() % 130;
 	for (int x = 10; x < 390; x+= 10) {
 		int y2 = rand() % 130;
 		sprintf(temp, "<line x1=\"%d\" y1=\"%d\" x2=\"%d\" y2=\"%d\" stroke-width=\"1\" />\n", x, 140 - y, x + 10, 140 - y2);
 		out += temp;
 		y = y2;
 	}
	out += "</g>\n</svg>\n";

	server.send ( 200, "image/svg+xml", out);
}
```

devyte, 2018-02-24:
Your problem is that delay. While delayed, the lwip stack can't hand off data to the server object, so buffers blow up. When serving, you can't use long delays. Instead, poll elapsed time with millis().
Closing as user error.

closed 2018-02-24 by devyte

s0170071, 2018-02-24:
I am not sure you got my point. That delay is not the problem, I am not even calling server.handleClient().
The LWIP is storing data until it crashes the ESP. It should stop before that happens, even if it means losing data. 
When my program attends to some sensors or assembling pages, it is vulnerable to DOS attacks as nothing is stopping the LWIP from allocating the memory if it is flooded with packets. 
You may call it a race condition or a bug. If memory is really low (happens with only 40k ;) you do not want to have many packets coming in...

igrr, 2018-02-24:
For each TCP connection, LWIP will only accept TCP_WND of data if the application does not read received data. By default that's 2xTCP_MSS, or 2920 bytes. That happens for each incoming connection. It's not a valid behavior for LWIP to drop data, but it can kill existing connections. I need to check the exact logic which is used, but I think at some point LWIP should start killing PCBs which are in "established" state. There's a limit on the number of these, so in theory we shouldn't be getting unbounded memory consumption. The limit should be something like 2920 bytes times the maximum number of PCBs (5?).
There's also an option to use smaller TCP_MSS of 536 bytes in the latest release, which can help in this case.