esp8266/Arduino #4429: Not able to upload code on my node mcu

qwerty12345asdf, 2018-02-24:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos
[image]

#### Hardware
Hardware:	NodeMCU DEVKIT 1.0		
Core Version:      	

### Description

Problem description:
Whenever i try to upload any code it shows me this 
warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

### Settings in IDE

Module:  		NodeMCU1.0(ESP-12E Module)	?Generic ESP8266 Module?
Flash Size: 	4 M(1M SPIFFS) 	?4MB/1MB?
CPU Frequency:	80Mhz
Flash Mode:	qio
Flash Frequency:40Mhz
Upload Using:	 SERIAL
Reset Method:	NODEMCU

### Sketch

`/*
 *  This sketch demonstrates how to scan WiFi networks. 
 *  The API is almost the same as with the WiFi Shield library, 
 *  the most obvious difference being the different file you need to include:
 */
#include "ESP8266WiFi.h"

void setup() {
  Serial.begin(115200);

  // Set WiFi to station mode and disconnect from an AP if it was previously connected
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  delay(100);

  Serial.println("Setup done");
}

void loop() {
  Serial.println("scan start");

  // WiFi.scanNetworks will return the number of networks found
  int n = WiFi.scanNetworks();
  Serial.println("scan done");
  if (n == 0)
    Serial.println("no networks found");
  else
  {
    Serial.print(n);
    Serial.println(" networks found");
    for (int i = 0; i < n; ++i)
    {
      // Print SSID and RSSI for each network found
      Serial.print(i + 1);
      Serial.print(": ");
      Serial.print(WiFi.SSID(i));
      Serial.print(" (");
      Serial.print(WiFi.RSSI(i));
      Serial.print(")");
      Serial.println((WiFi.encryptionType(i) == ENC_TYPE_NONE)?" ":"*");
      delay(10);
    }
  }
  Serial.println("");

  // Wait a bit before scanning again
  delay(5000);
}
```

### Debug Messages

```
messages here
```
any help would be really helpful?

RudyFiero, 2018-02-24:
Look at this.
http://arduino-esp8266.readthedocs.io/en/latest/faq/a01-espcomm_sync-failed.html

This is not the right place for this question. It has nothing to do with the code. Try www.esp8266.com/

closed 2018-02-24 by devyte

qwerty12345asdf, 2018-02-25:
thanks I was able to solve it 👍