esp8266/Arduino #4432: 8266 Wi-Fi Problem

pletchford, 2018-02-25:
#### Hardware
Hardware:		ESP 12E development board
Core Version:      	2.4.0-rc2

### Description
system working well locally on home network but will not work on a remote network.
Nothing to do with router or port forwarding as I have a few other boards working with the same router and network. Also have tried different router just in case.
Description:
A very unusual error for me anyway. Happens in both platform I/O and Arduino
the code compiles with no errors and runs locally within your home subnet for me that is 192.168.1.X
I have setup my router for Port forwarding and checked most of the obvious things.
One interesting aspect is if I run on an earlier version of the Arduino board manager 2.4.0 the program runs fine from a remote site if I upgrade to version 2.4.0 R2 program still runs locally fine ,but I cannot connect from a remote site.
Also additionally there is no error messages the code compiles fine uploads fine works locally by but remotely will not work with the latest version 2.4.0.r2 but again if you just take the board manager back to version 2.4.0 then works remotely and locally.
the full programme running can be found at meathome2017.hopto.org

Very strange I do not know where else to post if this is the incorrect area.
If anyone could show me what I’m doing wrong I would much appreciate it.
Thanking you for your help in advance
### Settings in IDE

Module:  			Generic ESP8266 Module?
Flash Size: 		4MB
CPU Frequency:       80Mhz?
Flash Mode:	         qio?
Flash Frequency:	?40Mhz?
Upload Using:		OTA / SERIAL? I tried using both it does not matter
Reset Method:		nodemcu

### Sketch

```cpp
#include <ArduinoOTA.h>
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>

const char* ssid = "BTHub5888888";
const char* password = "88888888";

ESP8266WebServer server(80);

const int led = LED_BUILTIN;

IPAddress ip(192, 168, 1, 202);  // ip of Client
IPAddress gateway(192,168,1,254);           // gateway of your network
IPAddress subnet(255,255,255,0);
IPAddress dns(192, 168, 1, 254);
unsigned long delays=2000;

String  theme = "darkly";
long Day=0;
int Hour =0;
int Minute=0;
int Second=0;
int HighMillis=0;
int Rollover=0;
String bmp_stv1,bmp_spv1,bmp_sav1;       //BMP sensor string values
float bmp_tv1,bmp_tpv1,bmp_tlv1,bmp_thv1; //BMP temperature sensors float values
float bmp_hv1,bmp_hpv1,bmp_hlv1,bmp_hhv1; //BMP humidity sensor float values
float bmp_av1,bmp_apv1,bmp_alv1,bmp_ahv1; //BMP atmospheric sensor float values
String dht_stv1,dht_shv1;               //DHT22 sensor string values
float dht_tv1,dht_tpv1,dht_tlv1,dht_thv1; //DHT22 temperature sensor float values
float dht_hv1,dht_hpv1,dht_hlv1,dht_hhv1; //DHT22 humidity sensor float values
String lightsv1,lightsv2,lightsv3,lightsv4,lightsv5;
int lightv1,lightv2,lightv3,lightv4,lightv5;
int lightpv1,lightpv2,lightpv3,lightpv4,lightpv5;
String  etatGpio[5] = {"HOME","Page2","Page3","Page4","Page5"};
const uint8_t GPIOPIN[5] = {1,2,3,4,5};
WiFiClient client;
[... 194 lines cut ...]
  page +=             "</span></button>";
  page +=             "<form action='/' method='POST'><button type='button submit'name='2' value='Page2' class='btn btn-primary'> <span class='badge'>";
  page +=             etatGpio[1];
  page +=             "</span></button>";
  page +=             "<form action='/' method='POST'><button type='button submit'name='3' value='Page3' class='btn btn-success'><span class='badge'>";
  page +=             etatGpio[2];
  page +=             "</span></button>";
  page +=             "<form action='/' method='POST'><button type='button submit'name='4' value='Page4' class='btn btn-warning'><span class='badge'>";
  page +=             etatGpio[3];
  page +=             "</span></button>";
  page +=             "<form action='/' method='POST'><button type='button submit'name='5' value='Page5' class='btn btn-danger'><span class='badge'>";
  page +=             etatGpio[4];
  page +=             "</span></button>";
  page +=             "</div><div>";
  return page;
}

```

### Debug Messages
no debug messages
```
messages here
compiles correctly
```

devyte, 2018-02-25:
2.4.0-rc2 was the second release candidate. It is older than release 2.4.0. Currently, release 2.4.0 is the latest release version.
Closing.

closed 2018-02-25 by devyte

pletchford, 2018-02-25:
This the older ver 2.4.0 the new ver will not work as I said.
so please can you open this back up as it will not work with the new ver

devyte, 2018-02-25:
you said:

> if I run on an earlier version of the Arduino board manager 2.4.0 the program runs fine from a remote site if I upgrade to version 2.4.0 R2 program still runs locally fine ,but I cannot connect from a remote site.

And:

> will not work with the latest version 2.4.0.r2 but again if you just take the board manager back to version 2.4.0 then works remotely and locally.

2.4.0-rc2 is older than 2.4.0.

pletchford, 2018-02-25:
Their was a version by the same name released before RC1 and its that version of the same name I am talking about not the latest version I don't know why the two had the same name very confusing all I can think it was a release candidate or something I have resubmitted as you have seen