esp8266/Arduino #4433: Arduino serial monitor - ESP8266-01 just works in loop() and does not in setup() ?

rugo100, 2018-02-25:
Hi
I am fighting with the below issue since a long time and therefore any suggestion would welcome.

I am using ESP8266-01
Upload of my sketch OK
Reboot, then my sketch is working perfectly (e.g. connects to WiFi, blinks LEDs etc) except Serial.print...
(for getting any clue  what my sketch is doing in time I print messages to serial port and I want to see them in IDE's Serial Monitor)
Situation is as follows: (port and a baud are set well and sketch uses Serial.begin(115200); in setup() )

1. if I start first Serial Monitor then hard reset ESP then the program is working well but nothing appear on serial monitor.
2. if I hard reset ESP first then go to Tools and start Serial Monitor too then things are going well, when any text sent to the serial port with Serial.print that appears on the monitor as it should be. So serial communication is working.

What the problem is? 
When my sketch starts in its setup() there is starting the serial communication (Serial.begin) and I do some setups -like connecting WiFi etc- also there are some Serial.print commands what would send out to the monitor some important information what I want to see. 
in above scenario 1. nothing appear on monitor
in above scenario 2. i "miss the time" up to that time when I can start the serial monitor from Tools menu the program is already over on setup() part what sends the important info what I would like to see.  
Putting e.g. delay(15000); into the sketch right after Serial.begin does not help...Seems to be serial communication just work if I start Serial Monitor (or any other "external" serial terminal program)  in the time when sketch is already running on loop() part...

I want to know
if Serial Monitor start first then secondly the sketch, in such a case why serial communication does not start/work ? (although the sketch is running well e.g. WiFi connection is established, LEDs are blinking etc)
Is that possible Serial communication just work in loop() part and does NOT in setup()?

If somebody can explain and even can suggest solution that would highly appreciated!

Also I would like to ask more experienced guys what is the difference betwenn IDE -Tools - Reset Mehod "ck" and "NodeMCU"  ? Is it something what I need to deal with connecting my above issue? 
(I have tried both, but I cannot see difference. ESP8266-01 is connecting my WINPC via FTDI board and only used RX,TX and GND) In IDE both related boards are tried (SparkFun ESP8266 Thing and generic ESP8266 too)

thx

rugo100, 2018-02-25:
To demonstrate my issu I put a test sketch to here.
I want to see text: "Test print from setup()" on serial monitor too, but I cannot.
If the sketch already running when I start the serial Monitor (or an external monitor) then I just can see text "nnnn print out text from loop()" 
--------------------------------------------
Would somebody please upload it to an ESP8266-01 and try it? (To see if the problem is just somewhere in my own HW/SW setup? (or it is more general -what I do not think so-)
Thanks

int i=0;
void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println("Test print from setup()");
  Serial.println("-----------------------");
  Serial.println();
 
}

void loop() {
  // put your main code here, to run repeatedly:
  i=i+1;
  Serial.print(i);
  Serial.println(" print out text from loop()");

}

devyte, 2018-02-25:
I believe you're supposed to put a small delay after Serial.begin(115200).

rugo100, 2018-02-25:
unfortunatelly small or even big delay does not work, situation is same. Now I am going to try lower baud...
thx

rugo100, 2018-02-25:
I have changed baud to 9600 but it does not help at all, situation is same.
Seems to be if terminal already started before the sketch starts then it cannot open the port to talk.
[image]
[image]

rugo100, 2018-02-25:
Upper picture is when first terminal started then the sketch. (Nothing comes to monitor)
Lower picture is captured when terminal started when sketch already was running.
On first pic you can see DCD and DSR (let whatever they are) are set which probably prevents to connect the device. (Just my 2 cents) And here comes the question of the day: what sets DCD and DSR ? The ESP is connected via FTDI board and only RX,TX and GND used. (Somebody who may knows well serial protocol can turn the light - thanks)

rugo100, 2018-02-26:
my next try was to add 'IF(Serial)' to wait until connection established

void setup() {
 Serial.begin(9600);
 while (!Serial) {
    ; // wait for serial port to connect.
 }
 Serial.println("Hello from setup...");
 Serial.println();
}

But no success.
Seems to be 'Serial' is set to true independently whether terminal is connected or not. For this to work mybe additional wires would needed.

So more consideration was needed
I was thinking the following happens: when we execute Serial.Begin the program counter goes on (just send command to serial interface to setup the connection, but does not wait until that happens, it continues with the next command immediately.
So until connection will established need time but during that time our program is running and executes the next commands.
Also since we use an 'asynchronous' communication probably also need time until talker and listener can 'pick up the line' find the first 'good' start bit, so the first few data sent maybe will lost.
Then -after connection is established, talker and listener are in synchron and first 'good' start bit found normal communication begins (but at this time our program is already somewhere inside deep in the loop() modul and doing the umpteen print.

So I wanted to "measure" the needed time between execution of 'Serial.begin' and the first well printed text on serial monitor.

To carry out this I wrote this sketch:
It will "measure" the needed time until good connection established. (The first number printed will show how much secs needed for connection)

int i=0;
void setup() {
  // put your setup code here, to run once:
Serial.begin(9600);
while (i<20) {
  Serial.print(i);
  Serial.print(" ");
  delay(1000);
  i=i+1;
}
i=0;
Serial.println();
Serial.println("Hello from setup()...");
Serial.println();
}

void loop() {
i=i+1;
Serial.print(i);
Serial.println(" text printed from loop()...");
delay(1000);
}

[image]

I have run above sketch several times and it shows consistently there is needed 7 secs until the first serial print will be OK. (I have repeated the test with 115200 baud too. In that case needed time was consequently 10 secs. Why? That can be a good question.)

Well, all of these means for me if I want to print out something useful right after Serial.begin() then I MUST wait at least for 10 secs then do the print.

So my own answer for the question asked in the title of the issue is: Yes ESP8266-01 serial printing is working in setup() of sketch too just we should give adequate time to establish the serial connection.

These checks were done with an "external" serial monitor (Tera Term) not with Arduino IDE's Serial Monitor.
Unfortunately above observations do not give answer for the question why Serial Monitor of Arduino IDE does not work if I start it first then I run my sketch. This question is still open for me. 

thx

devyte, 2018-02-27:
I didn't completely understand your issue before.
There are commercial boards, like the Wemos D1 mini or Nodemcu, that use one reset method. It is not the only reset method possible. We support those two reset methods for the hardware.
[This link](https://tttapa.github.io/ESP8266/Chap06%20-%20Uploading.html) may be relevant to you, or just google esp8266 reset method or whatever.

You said:
> what sets DCD and DSR ? The ESP is connected via FTDI board and only RX,TX and GND used.

Does the FTDI board have those pins at all? if they're NC, are they floating?

The timeout sounds like it could be some timeout. Maybe it's a driver issue, or an issue with your FTDI board.

In any case, I don't see how this is related to the core hosted in this repo. Please refer to a community forum for further assistance.
Closing as off-topic.

closed 2018-02-27 by devyte

rugo100, 2018-03-01:
All of my troubles were caused by used power supply!
After i changed to a good one everything started to work well

AndyUK1966, 2019-08-05:
Serial.begin(74880); // Works with the latest Arduino IDE, you'll see info from the ESP as it boots.
Serial.println(); // Don't forget this

carolesr, 2020-07-11:
> my next try was to add 'IF(Serial)' to wait until connection established
>
> void setup() {
> Serial.begin(9600);
> while (!Serial) {
> ; // wait for serial port to connect.
> }
> Serial.println("Hello from [...]

Okay you saved me a lot of time and pain, thank you!

Dooofeee, 2022-07-14:
> I believe you're supposed to put a small delay after Serial.begin(115200).

Worked for me, thanks!