esp8266/Arduino #4435: CI building for PlatformIO may have been semi-broken?

Adam5Wu, 2018-02-25:
I tried to reuse the CI building script in my own fork of this repo, and I enncountered a strange problem with the PlatformIO part, which I think may due to a subtle bug in the current script.

Since I am not actively using PlatformIO (plan to try soon), so I may be mistaken myself.

Anyway, here is the part of concern:
https://github.com/esp8266/Arduino/blob/cd6cd85fca3e883179bb757183ab36c594eb9940/tests/common.sh#L149-L150

Reading the code, it seems to try to make platformIO believe the "framework-arduinoespressif8266" component has already been installed as the current framework source being built.

However, this approach is ineffective at least in current version of PlatformIO because I later encountered error saying some file not found, but those files clearly exists in the repo being built.

Upon further investigating the error message, apparently PlatformIO is using a different directory name for the actual framework source: (note the **@1.20400.2** suffix)
```
Dependency Graph
|-- <ESP8266mDNS> (/home/travis/.platformio/packages/framework-arduinoespressif8266@1.20400.2/libraries/ESP8266mDNS)
...
xtensa-lx106-elf-g++ -o .pioenvs/nodemcuv2/src/SecureHTTPSUpdater.ino.cpp.o ... -I/home/travis/.platformio/packages/framework-arduinoespressif8266@1.20400.2/tools/sdk/include ...
```

So it seems the current PlatformIO CI build may be actually building the library examples using a **different (earlier released)** version of esp8266 Adurino core, instead of using the current one being built.

I further checked the PlatformIO documentation, and it seems it provided a slightly different way of pointing a repo to locale file: http://docs.platformio.org/en/latest/userguide/platforms/cmd_install.html#description

So I did the following change, and it seems the CI building in my fork works fine now:
```
-- sed -i 's/https:\/\/github\.com\/esp8266\/Arduino\.git/*/' ~/.platformio/platforms/espressif8266/platform.json
++ sed -i "s:https\://github\.com/esp8266/Arduino\.git:file\://$TRAVIS_BUILD_DIR:" ~/.platformio/platforms/espressif8266/platform.json
```

chrisaddams, 2018-03-01:
I'm having pretty much this exact issue with platform IO right now...

Adam5Wu, 2018-03-01:
@chrisaddams  can you confirm my fix at the bottom solve your problem?
If so maybe I am more confident to make a PR for it. 😄

ivankravets, 2018-03-07:
@Adam5Wu could you try this code?

```bash
function install_platformio()
{
    pip install --user -U https://github.com/platformio/platformio/archive/develop.zip
    platformio platform install https://github.com/platformio/platform-espressif8266.git#feature/stage --without-package framework
    sed -i 's/https:\/\/github\.com\/esp8266\/Arduino\.git/*/' ~/.platformio/platforms/espressif8266/platform.json
    ln -s $TRAVIS_BUILD_DIR ~/.platformio/packages/framework-arduinoespressif8266
    # Install dependencies:
    # - esp8266/examples/ConfigFile
    pio lib install ArduinoJson
}
```

Instead of https://github.com/Adam5Wu/Arduino-esp8266/blob/adam5wu/master/tests/common.sh#L145

Adam5Wu, 2018-03-07:
@ivankravets Unfortunately the build still fails with error:
```
esp8266/libraries/ESP8266HTTPUpdateServer/examples/SecureHTTPSUpdater/SecureHTTPSUpdater.ino ------------ 
pio ci /home/travis/build/Adam5Wu/Arduino-esp8266/libraries/ESP8266HTTPUpdateServer/examples/SecureHTTPSUpdater --board nodemcuv2 --verbose
/tmp/tmpFM20rh/src/SecureHTTPSUpdater.ino:47:36: fatal error: ESP8266WebServerSecure.h: No such file or directory
```
Relevant diagnostic info:
```
[Wed Mar  7 20:57:29 2018] Processing nodemcuv2 (platform: espressif8266; board: nodemcuv2; framework: arduino)
--------------------------------------------------------------------------------
PLATFORM: Espressif 8266 (Stage) > NodeMCU 1.0 (ESP-12E Module)
SYSTEM: ESP8266 80MHz 80KB RAM (4MB Flash)
xtensa-lx106-elf-g++ -o "/tmp/tmpFM20rh/src/SecureHTTPSUpdater.ino.cpp" -x c++ -fpreprocessed -dD -E "/tmp/tmpTpp5r7"
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(light)
Collected 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266HTTPUpdateServer> v1.0 (/home/travis/.platformio/packages/framework-arduinoespressif8266@1.20400.2/libraries/ESP8266HTTPUpdateServer)
|   |-- <ESP8266WebServer> v1.0 (/home/travis/.platformio/packages/framework-arduinoespressif8266@1.20400.2/libraries/ESP8266WebServer)
|   |   |-- <ESP8266WiFi> v1.0 (/home/travis/.platformio/packages/framework-arduinoespressif8266@1.20400.2/libraries/ESP8266WiFi)
|   |-- <ESP8266WiFi> v1.0 (/home/travis/.platformio/packages/framework-arduinoespressif8266@1.20400.2/libraries/ESP8266WiFi)
|-- <ESP8266mDNS> (/home/travis/.platformio/packages/framework-arduinoespressif8266@1.20400.2/libraries/ESP8266mDNS)
|   |-- <ESP8266WiFi> v1.0 (/home/travis/.platformio/packages/framework-arduinoespressif8266@1.20400.2/libraries/ESP8266WiFi)
|-- <ESP8266WiFi> v1.0 (/home/travis/.platformio/packages/framework-arduinoespressif8266@1.20400.2/libraries/ESP8266WiFi)
...

```

AsafFisher, 2018-03-25:
Any  solution to this peoblem?

Adam5Wu, 2018-03-25:
I confirm the proposed method at the end of the original post still works.
Maybe I will make a PR

AsafFisher, 2018-03-25:
Still gives me:
`#include <ESP8266WebServerSecure.h>
^
compilation terminated.
*** [.pioenvs/esp12e/src/main.cpp.o] Error 1`

AsafFisher, 2018-03-25:
Found a solution. Add to platformio.ini the git source.
Follow these steps:
http://docs.platformio.org/en/latest/platforms/espressif8266.html#using-arduino-framework-with-staging-version

Please mark as closed. Build works.

Adam5Wu, 2018-03-25:
@AsafFisher could you please give a bit more detail? Did you forward a build environment value into the platformio.ini?

AsafFisher, 2018-03-28:
Simply in the platformio.ini and set platform to

[env:nodemcuv2]
platform = https://github.com/platformio/platform-espressif8266.git#feature/stage
board = nodemcuv2
framework = arduino

Adam5Wu, 2018-03-28:
@AsafFisher this will instruct the PlatformIO to download from master branch of this repo.

But in the Travis CI context, it is not correct to do so. What should happen is that the **exact commit version** of esp8266 Arduino should be the one used by Platform IO.

Say you open a new branch and added a new method `scramble` in the String class, and an example to show to use the new method `scramble`. The CI build will never succeed in building the new example if it always download esp8266 Arduino from master branch.

It needs to use the local checkedout copy of the modified esp8266 Arduino, only that one have the new method `scramble`.

ivankravets, 2018-04-03:
@Adam5Wu please take a look at https://github.com/esp8266/Arduino/pull/4273#issuecomment-378320666

earlephilhower, 2019-01-16:
I believe we're running okay now (PIO builds fail in the same way as non-debug PIO builds, for example, when we bust something).  Closing, but reopen if it's still there.

closed 2019-01-16 by earlephilhower