esp8266/Arduino #4436: Broken ESP8266 Servo support?

Gargo, 2018-02-25:
I have Nodemcu and Wemos D1 R1 boards + sg90 servo.
According to most examples I just need to connect 3 wires to board and run any appropriate code example.
NOTHING works!

In the same time the same code and scheme works on usual arduino.

Could you explain what is wrong? Of course arduino has 5V pinouts but in the same time there are a lot of examples with workable Nodemcu (Wemos has similar hardware).

devyte, 2018-02-27:
@Gargo When you opened this issue you were presented with a template requesting specific info. You ignored it in its entirety. Please edit your post and add the info that was requested.

devyte, 2018-03-29:
Closing as can't reproduce, and lack of feedback in a month.

closed 2018-03-29 by devyte