esp8266/Arduino #4437: SDWebServer Example - Link in /edit/index.htm not working

PhilippRott, 2018-02-25:
### Description
If you try to edit files on the sdcard by espXX.local/edit, this will work only, if file is in root path like: "/test.txt"

If you create a folder on the sd-card "/folder1/test.txt" the link on tree-view is:
`<li id="/folder1/folder1/test.txt"> ...`

So, the path will be double and the link is not working - files can not be edited or displayed.

#### My solution to fix this problem:

edit line 378 in  Arduino/libraries/ESP8266WebServer/examples/SDWebServer/SdRoot/edit/index.htm
change this:    
`leaf.id = (((path == "/")?"":**path**)+"/"+name).toLowerCase();`
to following:    
`leaf.id = (((path == "/")?"":**""**)+"/"+name).toLowerCase();`

#### Hardware
Hardware:			ESP32, but expecting the same for esp8266, because its a html problem!

earlephilhower, 2018-02-25:
Would you like to submit a PR for your fix?  Seems like you've already got a failing and passing test case with it, and it'll get your name in the "git blame" for all time. :)

PhilippRott, 2018-03-01:
I never tried to submit a PR before, need to read how to do this in the correct way ;-)

earlephilhower, 2018-03-01:
If it's a problem I can do it with your info here, but it's not too complicated...there are official readmes, but below is my general flow:

Fork the repo from github website
````
git clone https://github.com/PhilippRott/Arduino
git checkout -b fixsdwebserver
<do changes, test>
git commit -a
git push (it'll give you the proper command line to push to a branch on your repo @ github)
````
Go back to your forked repo @ GH and there'll be a new button saying "Branch X pushed Y mins ago.  Click here to cubmit a pull request")

PhilippRott, 2018-03-01:
thanks for your help, my PR is now pending...

PhilippRott, 2018-03-02:
@earlephilhower seems to finished with a PR error - any idea what's the problem?

earlephilhower, 2018-03-02:
I just restarted it.  There was a build timeout (you can check the details link at the bottom), sometimes happens when there's heavy load on the CI system.

earlephilhower, 2019-01-22:
PR #4451 fixes, link

earlephilhower, 2019-01-23:
Related to #5525

devyte, 2019-10-29:
@earlephilhower what's the status here? I see the PR is ready, I assume it just needs testing?

devyte, 2019-11-06:
Pushing back after internal discussion

earlephilhower, 2019-12-09:
I've tested the existing code and am not able to reproduce any failures.  I've looked at pre-existing files in subdirs on the card, and all is well.  If I make /d1/d2/hello.txt, I can open it just find in the editor while navigating down on the LHS.  If I use the "mkdir" button to make a subdir and then the mkfile button is pressed, I also have no problem and the links on the LHS are all good.

Closing as can't repro...

closed 2019-12-09 by earlephilhower