esp8266/Arduino #4438: maybe a fix for WiFi.setAutoReconnect (false) never connecting

odelot, 2018-02-26:
Using: 2.4.0 git version

I always used WiFi.persistent (false); and took care of ssid and pass stored in the eeprom by myself. Until 2.3.0 everything was working. I was connection with begin function manually.

In 2.4.0 changed something and my sketch was just connecting once. The second one never got the WL_CONNECTED status.

When used WiFi.setAutoReconnect (false) I discovered that I wasn't manually reconnecting before. It was the device. With WiFi.setAutoReconnect (true) I got a error that my wifi network was not found and with WiFi.setAutoReconnect (false) it never connects after the first connection.

In file ESP8266WiFiSTA.cpp I changed this part of the begin function to set the config even if it hasn't changed if we are in the persist mode. Just this way I could kick off the connection procedure and DHCP 

```
wifi_station_get_config(&current_conf);
    if(sta_config_equal(current_conf, conf)) {
        DEBUGV("sta config unchanged");

		**if (!WiFi._persistent) {
			//set anyway
			wifi_station_set_config_current(&conf);
		}**
    }
    else {
        ETS_UART_INTR_DISABLE();
```

I tried to make a small sketch to simulate it but I couldn't. This behavior seems to happen just with my big project. 

I will try to simulate this in a smal sketch, maybe I am doing something wrong, calling a disconnect in a wrong place, I don't know. I just wanted to share this to someone that are facing the same problem.

devyte, 2018-02-27:
@odelot git has updated SDK to 2.2, where there is an incompatibility with the wifi settings. If you update your sketch from a git version before the sdk change to a git version after the sdk change, you can have weird connections issues. You have to erase the flash (or just the settings) after updating, and then do the wifi setup.
There is ongoing work to handle this automagically under the hood, but it's not done yet.

odelot, 2018-02-27:
@devyte I am using the most updated code (sdk 2.2) and I've used the new menu option to erase the sketch and wifi settings, but got a weird behavior anyway. Just with the code above that everything went normal.

devyte, 2018-02-27:
Does the same behavior happen with pr #3798 ? I think without that one persistent (or not persistent) may be incomplete,.

odelot, 2018-02-27:
@devyte I will try and let you know. Tks!

odelot, 2018-03-05:
@devyte it seems to work! If I got something different I let you know

closed 2018-03-05 by odelot