esp8266/Arduino #4445: Esp8266 Reconnection Problem Version 2.4.0

mustafayuce33, 2018-03-01:
### Basic Infos

#### Hardware
Hardware:			ESP-12
Core Version:      	2.4.0

### Description

With previous version of SDK, esp tries to reconnect wifi network when its connection is lost. But with 2.4.0 version it can not recognize connection state. I have tested both version. Esp with 2.3.0 version work well. 

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz?
Upload Using:		OTA
Reset Method:		ck

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

I have set Serial.setDebugOutput(true); and look at the logs. Here are the logs with 2.3.0 version ;

```
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.6.169
pm open,type:2 0
Invalid WEP key Aa123456
ap_loss
state: 5 -> 0 (0)
rm 0
pm close 7
reconnect
scandone
no ARMAKOM found, reconnect after 1s
reconnect
scandone
no ARMAKOM found, reconnect after 1s
reconnect
no ARMAKOM found, reconnect after 1s
reconnect
scandone
no ARMAKOM found, reconnect after 1s
reconnect
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with ARMAKOM, channel 6
dhcp client start...
ip:192.168.6.169,mask:255.255.255.0,gw:192.168.6.1
pm open,type:2 0

```
Here are the logs with 2.4.0 version ;

```
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Waiting for connection result with time out
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.6.169
*WM: freeing allocated params!

```

igrr, 2018-03-08:
Could you please

1. update the PR description to include the sketch used to reproduce this issue
2. install version 2.4.1 using boards manager and try again

mustafayuce33, 2018-03-08:
I have uploaded a piece of code. Here in our office when i reboot modem device , esp catches this issue and tries to reconnect again. But sometimes it doesnt. It stops somewhere in the code. When i look at the Wifi.Status() it looks WL_IDLE_STATUS. So that in my sketch i have added an timer that checks connection status and if it is not "connect" then i call "reconnect" function. 

And i also install latest git version by help of  lihttps://github.com/esp8266/Arduino#using-git-version link. This might cause this issue?

d-a-v, 2018-03-08:
Can you try with
```WiFi.setSleepMode(WIFI_NONE_SLEEP);```

tablatronix, 2018-03-08:
This sounds alot like and issue I was seeing also, where I know the ap was exists, but scandone and says nope. hmm

mustafayuce33, 2018-03-08:
Hi dave. Ok i will try your suggestion. And leave feedback again.

mustafayuce33, 2018-03-08:
Hi tablatronix

I am actually trying with theatest git version. It is not solved i think.

tablatronix, 2018-03-08:
You'll need a smaller minimal sketch to reproduce, one with no libraries.

tablatronix, 2018-03-08:
I could not reproduce with basic sketch and staging, tried all sorts of stuff. Did not try the example sketch there are too many libs and variables

mustafayuce33, 2018-03-09:
tablatronix , Thank you for your help. The example sketch is no so much different from basic sketch. I dont encounter this problem with some modems . Here at our office there is a modem brand name is "GrandStream". This modem causes this problem sometimes.

devyte, 2018-07-13:
There have been several fixes merged since this was opened, especially into the lwip glue. The current connection status has thoroughly tested. If there are problems connecting to a specific wifi router, the problem is either in the router, in which case there is nothing we can do, or in Espressif's SDK, in which case there is nothing we can do in this repo.
Closing.

closed 2018-07-13 by devyte