esp8266/Arduino #4447: UDP parsePacket erase previously packet even if there is no new packet

chiararuggeri, 2018-03-01:
#### Hardware
Hardware:			ESP8266
Core Version:      	2.4.0

### Description

I'm working with a board with an AVR328 as main chip and an ESP8266 as WiFi module. I would like that the ESP8266 continuously poll for WiFi events and notify them to the main chip but I'm having some issue working with UDP packets.
To check if a new UDP packet is coming I use UDP.parsePacket() function in an endless loop. The problem is that, when there is a new packet, the main chip is not quick enough to fetch it before a new call to UDP.parsePacket() deletes the old udp packet. parsePacket() function deletes the old packet even if there is no any new incoming packet. I temporarely fixed the issue by calling parsePacket() function only after 10 seconds if there is a new udp packet, but if a new packet arrives after 5 seconds I need to wait the whole time anyway before to be able to notify the main chip again.
Is there a way to know if there is a new incoming udp packet without erasing the previous one? If not, could you please modify the parsePacket() function in order to avoid erasing the old packet if there is no new incoming packet?

RudyFiero, 2018-03-02:
Use UDP.parsePacket() to read the packet into a buffer, and then set a flag for the AVR328 to indicate a packet is available. When you send the data to the AVR328 then clear that flag.

chiararuggeri, 2018-03-02:
I followed your suggestion with some little modifications and now things work better.
Thank you!

closed 2018-03-02 by chiararuggeri