esp8266/Arduino #4452: WiFi.softAP(ssid, password);  No change

Marcianikus, 2018-03-02:
----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			ESP-12
Core Version:                  2.4.0

### Description
Change "ssid, password" variables in code, upload, reset
and the module uses the old ssid and password, not recognize the new ssid and password, uses the old. 

### Settings in IDE

Module:  			Wemos D1 R2 mini
Flash Size: 		4MB/1MB
CPU Frequency:       80Mhz?
Flash Mode:			arduinoISP.org
Flash Frequency:	?40Mhz?
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch

```cpp

#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <WiFiClient.h> 
#include <ESP8266WebServer.h>
#include "TimerObject.h"

#define forward 0
#define backward 1
/* Set these to your desired credentials. */
const char *ssid = "Pruebas";
const char *password = "1234";

bool highBeamActive = false;
bool motionActive = false;
bool motionDirection = forward;
int target_speed;
int actual_speed;
int acceleration_step = 5;

int LED = 2; // Led D4
//int LED_BL1 = 13; // Red light on side 1 GPIO13
//int LED_HB2 = 14; // High beam side 2 GPIO12
//int LED_HB1 = 12; // High beam side 1 GPIO14

int LED_HDL = 2;  //PWM for headlights controlled by motor board

int motor_AIN1 = 13; //D7
int motor_AIN2 = 12; //D6
int motor_PWM =  14; //D5

TimerObject *timer1 = new TimerObject(10);

ESP8266WebServer server(80);

/************************************************************************** 
 *  
 *  Server IP address is: http://192.168.4.1
 *  
 *  When you have connected your device to the wifi network, open the browser and enter the IP address above.
 *  
 */
void handleRoot() {
  /************************************************************************ 
  * This is the web page sent to the connected device. There are lots of memory available so this page can be made much more complex. 
  * It uses Ajax to send commands which means the page doesn't need to be reloaded each time a button is pressed. In this way, 
  * the page can be made much more graphically complex without suffering from long reloading delays.
  */
  String html ="<html> ........................................, html)";
}

void motionControl(){
  if(motionActive){
    if(actual_speed < target_speed){
      actual_speed = actual_speed + acceleration_step;
    } else if(actual_speed > target_speed){
      actual_speed = actual_speed - acceleration_step;
    }    
    if(actual_speed > 1023){
      actual_speed = 1023;
    }
    if(actual_speed < 0){
      actual_speed = 0;
    }
    analogWrite(motor_PWM, actual_speed);
  }  else {
    analogWrite(motor_PWM, 0);
  }
}

/**************************************************
 *  Motor operation
 */

void motorOperation(){

  motionActive = true;  
  String move_dir = server.arg("dir");
  String motion_speed = server.arg("speed");
  target_speed = motion_speed.toInt();  

  //Execute change of direction, but only if actual_speed is below 100, otherwise set target speed to 0 and initiate a slowdown.  
  if(move_dir == "backward"){
    if(motionDirection != backward && actual_speed > 100){
      target_speed = 0;
    } else {
      motionDirection = backward;
    }
  } else {
    if(motionDirection != forward && actual_speed > 100){
      target_speed = 0;
    } else {
      motionDirection = forward;
    }
  }

  digitalWrite(motor_AIN1, motionDirection);
  digitalWrite(motor_AIN2, !motionDirection);
  //digitalWrite(LED_BL2, motionDirection);
  //digitalWrite(LED_BL1, !motionDirection);
    
  //if(highBeamActive){
    //digitalWrite(LED_HB2, motionDirection);
    //digitalWrite(LED_HB1, !motionDirection);
 // } else {
    //digitalWrite(LED_HB2, 0);
    //digitalWrite(LED_HB1, 0);
 // }

  
}

void updateSpeed(){
  String motion_speed = server.arg("speed");
  target_speed = motion_speed.toInt();  
}

void switchLightOn(){
    highBeamActive = true;
    digitalWrite(LED, 1);
    //digitalWrite(LED_HB1, !motionDirection);
}

void switchLightOff(){
    highBeamActive = false;
    digitalWrite(LED, 0);
    //digitalWrite(LED_HB1, 0);  
}

void motionStop(){
  motionActive = false;
  analogWrite(motor_PWM, 0);    
}

/*********************************************************************
 * SETUP
 */

void setup() {
	delay(1000);
  Serial.begin(115200);

  pinMode(LED, OUTPUT);
 // pinMode(LED_BL1, OUTPUT);
 // pinMode(LED_HB2, OUTPUT);
 // pinMode(LED_HB1, OUTPUT);
  
 // pinMode(LED_HDL, OUTPUT);
  
  pinMode(motor_AIN1, OUTPUT);
  pinMode(motor_AIN2, OUTPUT);
  pinMode(motor_PWM, OUTPUT);

  
  //digitalWrite(LED_BL2, 0);
  //digitalWrite(LED_BL1, 0);
  //digitalWrite(LED_HB2, 0);
  //digitalWrite(LED_HB1, 0);

  digitalWrite(motor_AIN1, 0);
  digitalWrite(motor_AIN2, 1);

  analogWrite(motor_PWM, 0);
  //analogWrite(LED_HDL, 1023);

  timer1->setOnTimer(&motionControl);
  timer1->Start();
  
	WiFi.softAP(ssid, password);

	//IPAddress myIP = WiFi.softAPIP();

 /****************************************************************************
  * Here you find the functions that are called when the browser sends commands. 
  */
	server.on("/", handleRoot);
  server.on("/run", motorOperation);
  server.on("/stop", motionStop);
  server.on("/lighton", switchLightOn);
  server.on("/lightoff", switchLightOff);
  server.on("/updateSpeed", updateSpeed);
	server.begin();
}

void loop() {
	server.handleClient();
  timer1->Update();
  Serial.println(ssid);
  Serial.println(password);

}
### Debug Messages

```
messages here
```

RudyFiero, 2018-03-02:
> Debug Msg: [AP] fail passphrase to long or short! wifi evt: 7

Read the Debug message and think to yourself, maybe this is a clue.

1234  Does not equal 8 characters. It rejected it because it is too short.

devyte, 2018-03-02:
Closing due to user error.

closed 2018-03-02 by devyte

Marcianikus, 2018-03-03:
old submit code:

const char *ssid = "gtgtgtgt";
const char *password = "1234";

actual submit code:

const char *ssid = "lalalala";
const char *password = "1234";

**ssid in movil and tablet:* gtgtgtgt

*WiFi.printDiag(Serial);*:

        Mode: AP
        PHY mode: N
        Channel: 1
        AP id: 0
        Status: 255
        Auto connect: 0
        SSID (0): 
        Passphrase (0): 
        BSSID set: 0

        wifi evt: 7

please taste the code,  is a clue.

:-)

Marcianikus, 2018-03-03:
You were right, but it is not documented that the password must be a minimum of 8 characters. It would be a good idea to add this to the documentation

Pablo2048, 2018-03-03:
IMHO it IS documented actually - see https://en.wikipedia.org/wiki/Wi-Fi_Protected_Access - You can't blame tools if you have no surrounding knowledge. Same is valid for valid characters in passwords...

nsisodiya, 2018-09-03:
Ok, dumb question.
I see lot of password-less IOT device. they create wifi network without any password. Click and join. How they do it if the minimum 8 characters limit present.
Or may I am asking wrong question.

DAVIDBETAN2000, 2019-05-30:
password must to have 8 char minimun to work