esp8266/Arduino #4453: WiFi can't connect after disconnecting (only lwIP v2)

JiriBilek, 2018-03-03:
### Basic Infos

Hi all, thanks for a excellent work an libraries for ESP8266.
Maybe, I found an issue in lwIP v2. I tried to search the internet and this repo and couldn't find a solution so I hope it is not a duplicate issue.

#### Hardware
Hardware:			ESP-12E
Core Version:      	2.4.0

### Description

No connection possible after WiFi.disconnect. After disconnecting and calling WiFi.begin(char*,char*) again, the WiFi.status() returns WL_DISCONNECTED forever.
This behaviour I observed on lwIP v2 only, on the older v1.4 the sketch runs fine.
And moreover, when I put the ESP8266 modem to sleep with WiFi.forceSleepBegin(), the sketch runs fine even on lwIP v2.

More in paragraph Debug Messages below.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/1MB SPIFFS
CPU Frequency:		80Mhz
Flash Mode:			QIO
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck
lwIP Variant:        v2 failing, v1.4 works fine

### Sketch

```cpp

#include <ESP8266WiFi.h>

const char* ssid     = "***";
const char* password = "***";

WiFiClient wifiClient;

void setup(void) {
    Serial.begin(115200);  // Start serial monitor at 115200 baud
    delay(100);

    WiFi.setAutoConnect(false);
    wifiConnect();

    delay(5000);

    Serial.println(F("Disconnecting..."));
    WiFi.disconnect();
//+    WiFi.forceSleepBegin();  // turn off ESP8266 RF

    delay(5000);
    
//+    WiFi.forceSleepWake();
    wifiConnect();
}

void wifiConnect(void) {
    WiFi.mode(WIFI_STA); 
    WiFi.begin(ssid, password);
        
    Serial.print(F("Connecting to AP "));
  
    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
        Serial.print(WiFi.status());
    }

    Serial.print(F("\nWiFi connected\nIP address: "));
    Serial.println(WiFi.localIP());
}

void loop(void) {}
```

### Debug Messages

In the debug messages there is intermixed my output with debugging output but I hope it would be clear.

#### Test 1 (lwIP v2, no modem sleep)

Setup: lwIP v2 Prebuilt (MSS=536), the sketch as-is
Result: stays forever in loop waiting for WiFi.status() == WL_CONNECTED

```
Connecting to AP wifi evt: 2
wifi evt: 0
.6.6wifi evt: 3
.3
WiFi connected
IP address: 192.168.1.102
Disconnecting...
wifi evt: 1
STA disconnect: 8
Connecting to AP .6.6.6.6.6wifi evt: 0
.6.6.6.6.6.6.6.6.6.6.6.6.6.6.6.6.6.6.6.6.6.6.6.6.6.6.6.6.6.6.6.6
```
#### Test 2 (lwIP v1.4, no modem sleep)

Setup: lwIP v1.4 Prebuilt, the sketch as-is
Result: runs fine

```
Connecting to AP wifi evt: 2
wifi evt: 0
.6.6.6wifi evt: 3
.3
WiFi connected
IP address: 192.168.1.102
Disconnecting...
wifi evt: 1
STA disconnect: 8
wifi evt: 8
Connecting to AP wifi evt: 8
.6.6.6.6.6wifi evt: 0
.6.6.6wifi evt: 3
.3
WiFi connected
IP address: 192.168.1.102
pm open,type:2 0

```
#### Test 3 (lwIP v2, modem sleep)

Setup: lwIP v2 Prebuilt (MSS=536), in the sketch I uncommented two lines causing modem sleep and wakeup (marked with //+ )
Result: runs fine, the modem goes off (the consumption dropped to 15 mA) and on (70 mA again)

```
Connecting to AP wifi evt: 2
.6.6wifi evt: 0
.6.6wifi evt: 3
.3
WiFi connected
IP address: 192.168.1.102
Disconnecting...
wifi evt: 1
STA disconnect: 8
wifi evt: 8
Connecting to AP wifi evt: 3
wifi evt: 8
.6.6.6.6.6wifi evt: 0
.6.6.6.6wifi evt: 3
.3
WiFi connected
IP address: 192.168.1.102
pm open,type:2 0
```

d-a-v, 2018-03-03:
Please retry with latest git master, this bug has been fixed since 2.4.0.

JiriBilek, 2018-03-03:
@d-a-v , thanks for the advice. After installing the latest git version, the issue is gone. I was using the "stable" 2.4.0 as I am generally against making permanent changes in my development environment.
Thanks.

closed 2018-03-03 by JiriBilek