esp8266/Arduino #4455: "pm open,type:2 0" output on Serial with Debug Level 'None' on udp.begin()

liebman, 2018-03-03:
### Basic Infos

#### Hardware
Hardware:			ESP-12f
Core Version:      	current git

### Description

Problem description

With Debug Level set to 'None' a message is printed on Serial when the begin() method of WiFiUDP is called. Grepping the for the message points to libpp.a

```
zod:esp8266 chris.l$ grep -r "pm open,type" .
Binary file ./tools/sdk/lib/libpp.a matches
```

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/1MB
CPU Frequency:	80Mhz
Flash Mode:		qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch

```cpp
/*
 *  This sketch sends data via HTTP GET requests to data.sparkfun.com service.
 *
 *  You need to get streamId and privateKey at data.sparkfun.com and paste them
 *  below. Or just customize this script to talk to other HTTP servers.
 *
 */

#include <ESP8266WiFi.h>
#include <WiFiUdp.h>

const char* ssid     = "XXXXXX";
const char* password = "XXXXXX";

WiFiUDP udp;

void setup() {
  Serial.begin(115200);
  delay(10);

  // We start by connecting to a WiFi network

  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  
  /* Explicitly set the ESP8266 to be a WiFi-client, otherwise, it by default,
     would try to act as both a client and an access-point and could cause
     network-issues with your other WiFi-devices on your WiFi-network. */
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  udp.begin(1234);
}

void loop() {
  delay(10000);
}

```

### Debug Messages

```
tail 8
chksum 0x2d
csum 0x2d
va2e7c7d0
~ld

Connecting to XXXXXX
.
WiFi connected
IP address: 
192.168.0.17
pm open,type:2 0

```

cprasmu, 2018-03-04:
Same issue here too 

wifi evt: 0
....wifi evt: 3
.
WiFi connected
IP address: 
192.168.1.110
pm open,type:2 0

igrr, 2018-03-04:
This is output from the SDK, which gets enabled whenever you set 'Debug Port' to something other than 'None', or call `Serial.setDebugOutput(true);` from the sketch.

liebman, 2018-03-04:
Thanks @igrr - good to know!

closed 2018-03-04 by liebman

regpa, 2018-06-17:
Dear @igrr, could you please elaborate about the meaning to this SDK output: "pm open,type:2 0" ?

igrr, 2018-06-17:
@regpa Sorry, can't tell for sure — it comes from the closed source part.
I *assume* it is related to sleep mode ("pm", "power management") being enabled ("open"), in which case "2" in "type:2" should match the selected sleep mode.