esp8266/Arduino #4456: WiFiClientSecure connect failed when call using os_timer

vlad2005, 2018-03-03:
### Basic Infos

#### Hardware
Hardware:			ESP-12E
Core Version:      	platformio with stable esp8266 arduino framework

### Description
Connection failed when call WiFiClientSecure->connect using os_timer  

#### code concept. 
```C

os_timer_t timer;
WiFiClientSecure client = new WiFiClientSecure();

void timerCallback(void *arg){
  client->connect("google host", 443);
}

setup(){
  os_timer_setfn(&timer, timerCallback, NULL);
  os_timer_arm(&timer, 10000, true);
}

loop(){
}
 
```
I have no problem with my code when i use SimpleTimer library to call timerCallback.

mtiutiu, 2018-03-04:
From the official documentation regarding the **Ticker** library which uses the **os_timer** functions behind the scenes:
`It is currently not recommended to do blocking IO operations (network, serial, file) from Ticker callback functions. Instead, set a flag inside the ticker callback and check for that flag inside the loop function` like the connect method from the

So I think this may be an issue because you're doing blocking operations inside a os_timer function callback like the connect method from the WiFiClientSecure class?

vlad2005, 2018-03-04:
Yes, SimpleTimer is based on millis(). Is an implementation of basic "blink withoud delay" concept.
Thanks!
I will try to change my code to use an flag. I will post the result later.

vlad2005, 2018-03-04:
@mtiutiu
Thanks again for suggestion! It's working nice.

closed 2018-03-04 by devyte