esp8266/Arduino #4458: delay() causes memory corruption

mcr-ksh, 2018-03-04:
### Basic Infos

#### Hardware
Hardware:			ESP-8266 / WeMos D1 mini
Core Version:      	
setup(): CHIPID: da78a4 VCC: 65535
setup(): FLASHID: 1640e0 FLASH SIZE: 4194304
setup(): FLASH SPEED: 40000000 MODE: DIO
setup(): SDK Version: 2.1.0(deb1901)

Archiving built core (caching) in: 
Sketch uses 422655 bytes (40%) of program storage space. Maximum is 1044464 bytes.
Global variables use 66660 bytes (81%) of dynamic memory, leaving 15260 bytes for local variables. Maximum is 81920 bytes.
Low memory available, stability problems may occur.

### Description

somehow delay(5000) with 5000 causes memory corruption.
I have a HTTP request where the struct is from a ArduinoJSON object.
thus kwr->cmd points to a string that was read from the HTTP socket.
in the below debug message you can see the command ist "noop". after the delay
the memory changed. 

### Settings in IDE

Module:  			Wemos D1 mini ESP8266 
Flash Size: 		       4MB/1MB
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	        no idea
Upload Using:		SERIAL
Reset Method:		button

### Sketch

```
  Serial.println("cmd_noop(): Sleeping %d msec.\n", milliseconds);
  Serial.println("XXX(): %x COMMAND: %s\n", kwr, kwr->cmd);      
  delay(milliseconds); // delay causes memory corruption
  Serial.println("XXX(): %x COMMAND: %s\n", kwr, kwr->cmd);      
```

### Debug Messages

```
cmd_noop(): -->
cmd_noop(): p1 = 5000
cmd_noop(): Sleeping 5000 msec.
XXX(): 3fffa0c4 COMMAND: noop
pm open,type:2 0
XXX(): 3fffa0c4 COMMAND: <|=|x|y|z|{|||}|00

JUST A GUESS: weird enough the esp seems to read and store broadcast msg in its memory. this is what might overwrite the memory. here my AppleTV 
XXX(): 3fff9fb4 COMMAND: noop
XXX(): 3fff9fb4 COMMAND: et=0,3,5ft=0x5A7FFFF7,0x55FDEsf=0x244md=0,1,2am=AppleTV5,3Cpk=42d7........................................................................624tp=UDPvn=65537    vs=356.19    ov=11.2.6vv=2⸮K

HERE my HUE upnp is picked up.
XXX(): --> kwr = 3fff9efc cmd = 55.250:1900
CACHE-CONTROL: max-age=100
LOCATION: http://192.168.x.x:80/description.xml
SERVER: Linux/3.14.0 UPnP/1.0 IpBridge/1.23.0
NTS: ssdp:alive
hue-bridgeid: 001788FFFE40DF71
NT: urn:schemas-upnp-org:device:basic:1
USN: uuid:2f4.....0-....-....-....-..............

71%⸮A⸮%⸮T%⸮⸮⸮
```

devyte, 2018-03-04:
@mcr-ksh please provide a Minimal Complete Verifiable Example sketch that reproduces your issue. Without it, this won't be looked at.

tablatronix, 2018-03-08:
how are you initializing your struct, and I am assuming you are using it in a callback maybe?

halogens, 2018-05-10:
there is not much to go on with this info so all i can say is you can use millis() function with a loop instead of  delay().  it will solve your problem.

devyte, 2018-09-06:
Closing due to lack of feedback.

closed 2018-09-06 by devyte