esp8266/Arduino #4460: Number of Max Concurrrent Connection in Station mode?

dannybres, 2018-03-04:
Hello,

I am using ESP8266WiFi in station mode to connect to my AP at home. It is controlling 3 RGB leds sets  via the GPIO (9 channels for 3 sets). There is an situation where the server that controls them all polls the ESP8266WiFi server for the status of the lights with many concurrent http requests, e.g.

 - /rgb1/status
 - /rgb1/colour
 - /rgb1/brightness
 - /rgb2/status
 - /rgb2/colour
 - /rgb2/brightness
 - /rgb3/status
 - /rgb3/colour
 - /rgb3/brightness

My code is below and as new connections are created on the server, I ass them to the array `clients`, once I have received a whole request and attempt to write back to that client, the line `clients[i]->println("HTTP/1.1 200 OK");` takes a long time to execute and return 0, as I do not believe it still has that connection open. 

**Is there a maximum number of connections in Station mode? 
Can I increase this?**

Thanks,

Daniel

#include <ESP8266WiFi.h>

const char* location = "tv room LED controller";
const int locationID = 2;

int whiteLedPin[3] = {5,14,3}; // Pin 8
int redLedPin[3] = {4,12,1}; // Pin 6
int greenLedPin[3] = {0,13,9}; // Pin 7
int blueLedPin[3] = {2,15,10}; // Pin 8

int rgbLedStatus[3] = {0,0,0};
String rgbColour[3] = {"FFFFFF","FFFFFF","FFFFFF"};

int WWLedStatus[3] = {0,0,0};
int ledWWBrightness[3] = {100,100,100};
#define MAX_CLIENTS 10
#define MAX_LINE_LENGTH 50
// Create an instance of the server
// specify the port to listen on as an argument
WiFiServer server(80);
WiFiClient *clients[MAX_CLIENTS] = { NULL };
String st[MAX_CLIENTS];

void printJsonForTemperatureAndHumidityToClient(WiFiClient client,int locationID) {
  client.print("{\"room\": \"");
  client.print(location);
  client.print("\",\"id\": ");
  client.print(locationID);
  client.print(", \"led status\": ");
  client.print(rgbLedStatus[0]);
  client.print(", \"rgb colour\": \"");
  client.print(rgbColour[0]);
  client.print("\"}");
}

unsigned int hexToDec(String hexString) {
  unsigned int decValue = 0;
  int nextInt;
  
  for (int i = 0; i < hexString.length(); i++) {
    
    nextInt = int(hexString.charAt(i));
    if (nextInt >= 48 && nextInt <= 57) nextInt = map(nextInt, 48, 57, 0, 9);
    if (nextInt >= 65 && nextInt <= 70) nextInt = map(nextInt, 65, 70, 10, 15);
    if (nextInt >= 97 && nextInt <= 102) nextInt = map(nextInt, 97, 102, 10, 15);
    nextInt = constrain(nextInt, 0, 15);
    
[... 183 lines cut ...]

        // If you want to disconnect the client here, then do this:
        clients[i]->stop();
        delete clients[i];
        clients[i] = NULL;
        delay(1);
        Serial.println("Client disconnected");
        Serial.println("");

      } else {
        Serial.print("Building request for client:");
        Serial.print(i);
        Serial.print(":");
        
        Serial.println(st[i]);
        // Add it to the string
        st[i].concat(newChar);
                // IMPORTANT: Nothing stops this from overrunning the string and
        //            trashing your memory. You SHOULD guard against this.
        //            But I'm not going to do all your work for you :-)
      }
    }
  }

}

d-a-v, 2018-03-05:
You can try to use the OOM debug option, call WiFiClient::flush() before ::stop() and for a sanity check, print a notice when your loop does not find a free client to hold the newcomer.
We currently have a max of 4 concurrent connections (lwip's MEMP_NUM_RAW_PCB). It means that the 5th will be waiting for one of the 4 to be closed before beeing accepted, but it won't be lost.
You may need to use tcpCleanup() too (check the faq).

dannybres, 2018-03-05:
What is meant by OOM debug?

> your loop does not find a free client to hold the newcomer.

How do I know that there is a newcomer? If I have 4 connections open with atleast one more waiting and call 
`WiFiClient newClient = server.available(); `

will the `newClient` close one of the open connections and open a new one? Should I make sure I only call `WiFiClient newClient = server.available();` if I have less than 4 connections open?

Thanks.

d-a-v, 2018-03-05:
I was wrong, I am currently running 12 simultaneous continuous tcp connections. I guess I need to understand the meaning of lwip's MEMP_NUM_*_PCB macros.

You get a newcomer if server.available() returns a valid client. It will not close your current active clients. I guess if lwip can't handle a new incoming client, then this client will wait until an already-opened one is closed.
I currently see nothing very bad with your sketch. Check your free heap, use OOM debug option, use tcpCleanup since you have frequent short connections. Your storing loop does nothing if there is no free index for newcomers.

dannybres, 2018-03-05:
Thanks, what is OOM debug?

d-a-v, 2018-03-05:
OOM = Out Of Memory, it's an option in ```tools>debug level``` that you should use with ```tools>debug port>serial```.
It will yell if malloc() returns null.

devyte, 2018-03-06:
This is not an issue in the core, but rather a "How do I..." type question. 
Closing per issue POLICY doc.

closed 2018-03-06 by devyte