esp8266/Arduino #4463: Serial.end() not working to free Tx / Rx pins

horendus, 2018-03-05:
### Basic Infos

#### Hardware
Hardware:			ESP-12
Core Version:      	2.40

### Description

Calling Serial.end() does not then allow you to use serial GPIO pins for other functions such as softwareSerial

### Settings in IDE

Module:  			NodeMCU
Flash Size: 		4mb
CPU Frequency:	120Mhz
Flash Mode:		NA
Flash Frequency:	?40Mhz?
Upload Using:		SERIAL?
Reset Method:		NA

I'm trying to use use the Rx and Tx pins for other purposes dynamically within my sketch (softSerial) however when I call serial.end() the pins do not seem to be re usable. I was under the impression the purpose of Serial.end() was to free those pins up during a sketch for other use. 

Basic Flow
- Call Serial.begin() & print to serial
- Call Serial.end()
- Call softwareSerial.begin() & print to softwareSerial
- Call softwareSerial.end()

Basically you cannot use softwareSerial on the same pins as hardwareSerial UNLESS you never called Serial.begin() within your sketch. There is no compile errors, it just doesn't work.

This make me think that in general the Serial.end() function is not working correctly and freeing up those pins 

The testing sketch looks something like this. The softwareSerial prints do not work. If you move the Serial.begin() to AFTER the first softwareSerial.begin() you can get one print to work but as soon as a Serial.begin() is called you no longer can use the pins for softwareSerial

 void setup(){
     Serial.begin(9600);
 {
  
void loop(){
  Serial.println("Hello Hardware Serial");
  delay(100);
  Serial.end();
  delay(100);
  softwareSerial.begin(9600);
  delay(100);
  softwareSerial.println("Hello Software Serial");
  delay(100);
  softwareSerial.end();
  delay(100);
  Serial.begin(9600);
}

VictorTagayun, 2018-12-17:
Hello,
I also see also garbage after reset pin put/tied low then reboots.

earlephilhower, 2019-01-24:
You haven't posted an MCVE, however this is pretty straightforward.

SoftwareSerial only sets `pinMode()` on object creation, not on `begin()`.  The serial port is setting the pins to `INPUT` when you call `end()` which is correct.  So you're reading the output of an input pin w/nobody driving it, so you're getting noise.

You'll either have to create the SWSerial object only after the HWSerial has `end()`ed, or open an issue in the SWSerial repo to see if he'd consider also setting the pinMode in SWSerial's `begin()` method.

closed 2019-01-24 by earlephilhower

pmcclay, 2020-03-15:
Just connecting dots:
This was addressed in the "official" SoftwareSerial a while ago
arduino/Arduino@8c8b78ce4975103ed82067fe98f2101fcce48d68