esp8266/Arduino #4464: python get.py fails on macOS

Paraphraser, 2018-03-05:
Following the setup instructions on macOS 10.12.6 (Sierra) results in a protocol error. A log of the terminal commands and output follows.

$ python --version
Python 2.7.10
$ pwd
~/Documents/Arduino/hardware/esp8266com
$ git clone https://github.com/esp8266/Arduino.git esp8266
Cloning into 'esp8266'...
remote: Counting objects: 16660, done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 16660 (delta 0), reused 0 (delta 0), pack-reused 16656
Receiving objects: 100% (16660/16660), 40.24 MiB | 1013.00 KiB/s, done.
Resolving deltas: 100% (9708/9708), done.
$ cd esp8266/tools/
$ python get.py 
Platform: x86_64-apple-darwin
Downloading esptool-0.4.13-osx.tar.gz
Traceback (most recent call last):
  File "get.py", line 116, in <module>
    get_tool(tool)
  File "get.py", line 75, in get_tool
    urlretrieve(url, local_path, report_progress)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib.py", line 98, in urlretrieve
    return opener.retrieve(url, filename, reporthook, data)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib.py", line 245, in retrieve
    fp = self.open(url, data)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib.py", line 213, in open
    return getattr(self, name)(url)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib.py", line 443, in open_https
    h.endheaders(data)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 1049, in endheaders
    self._send_output(message_body)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 893, in _send_output
    self.send(msg)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 855, in send
    self.connect()
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 1274, in connect
    server_hostname=server_hostname)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/ssl.py", line 352, in wrap_socket
    _context=self)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/ssl.py", line 579, in __init__
    self.do_handshake()
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/ssl.py", line 808, in do_handshake
    self._sslobj.do_handshake()
IOError: [Errno socket error] [SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:590)
$

Paraphraser, 2018-03-05:
This pointed me in the right direction:
`https://stackoverflow.com/questions/44316292/ssl-sslerror-tlsv1-alert-protocol-version`
The following was sufficient to overcome the problem:
```
brew update
brew install openssl
brew install python3

```
which upgraded Python to:
```
$ python --version
Python 3.6.4

```
After that, "python get.py" ran without issue.

CarlosRDomin, 2018-03-07:
Yeah, some of my friends had the same issue with Python 2.7. It seems to be an OpenSSL issue in general (not Python3 specific I mean). They followed [a different StackOverflow post](https://stackoverflow.com/questions/24323858/python-referencing-old-ssl-version/46308535#46308535) but the idea was the same, installing the latest openssl and Python through `Homebrew` (they installed Python 2 and it also fixed the issue).
Any idea of whether/how the script could be modified to also work with the default Python 2/3 on Mac? Or do we need to rely on Homebrew?

Paraphraser, 2018-03-07:
I started by fiddling with get.py but I stopped. I didn't have the resources to test all the platform dependencies and I was leery of introducing any Mac-specific dependency.

Seems to me that if HomeBrew can fix the problem without needing any other mods then that's the simplest way forward for everyone.

dalbert2, 2018-04-10:
The same issue exists on Windows 10 (64-bit) where py.get fails with the same protocol version error when using Python 2.7 but works when using Python 3.x

earlephilhower, 2019-05-01:
Closing this one as wontfix, same reasoning as in #4726 as it's SSL trusted CA related or SSL version related (so out of our hands, that's the OS' job).  I think GH now has a MSFT cert and server setup, so things have changed significantly on their end as well.

closed 2019-05-01 by earlephilhower