esp8266/Arduino #4466: ESP8266 client.state=-3

Hokahonay, 2018-03-05:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:		ESP8266-01
Core Version:      	dont know

developing IOT for home use using MQTT all library's are current
i have several modems in different locations all of which need to be able to connect to any node, because of this the WiFi connect logic cycles through the modem list until it finds one to connect to

Problem description
programme will work well for a length of time (minutes, not hours) then the client suddenly looses connection. the usual error code returned from client.state() is -3 although sometimes it is -2, i believe both of these error codes are WiFi related although i have not been able to find a definitive list of codes.
after the client disconnect one of several things happen:

1) the client reconnects and continues
2) the WiFi connection is lost and it reconnects and continues
3) the WiFi connection is lost and the connection logic cycles through all of the modems several times before eventually re connecting/

### Settings in IDE

Module:  			        Generic ESP8266 Module
Flash Size: 		        4M/3M SPIFFS
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	        40Mhz
Upload Using:		        SERIAL
Reset Method:		        ck

### Sketch
#include <ESP8266WiFi.h>
#include <PubSubClient.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>
#include <Streaming.h>
#include <EEPROM.h>
#include <SimpleDHT.h>
#include <Ticker.h>

WiFiClient espClient;
PubSubClient client(espClient);
SimpleDHT22 dht22;
Ticker MyTimer;
//=================================================
// variables
uint8_t modemNo;
bool    ModemFound = false;
char    MyConnection = 'N';
bool    MqttConnected = false;

#define EEValid  0
#define ModemLoc 10

// Modems & logon parameters
String ssidList[7] = {"modem1", "modem2", "modem3", "modem4", "modem5"};
String pwList[7]   = {"pw1", "pw2", "pw3", "pw4", "pw5"};
String nodesModems = {"modem1,modem2,modem3,modem4,modem5"};

char*               serverIP    = "192.168.1.19";
const unsigned int  serverPort  = 1883;
const char*         userID      = "user";
const char*         userPw      = "userpw";
const char*         MyMqttId    = "mqttuser";
//=================================================
void waitSecs(int secs) {
  long tmr = millis() + (secs * 1000);
  while(millis() < tmr) yield();
}
//=================================================
void doSubscribe() {
  String strg;
  // subscribe to topics
  strg = "LIGHTS/#";
  client.subscribe(strg.c_str());
  strg = "HEATING/#";
  client.subscribe(strg.c_str());
  client.setCallback(mqttMessages);
}
//=================================================
void wifiConnect() {
  char ssBuf[25]; char pwBuf[25]; // buffers for ssid & pw
  String strg;  uint8_t trys;
  WiFi.mode(WIFI_STA);
  while (WiFi.status() != WL_CONNECTED) {
    strg = String(ssidList[modemNo]); // get current modem
    strg.toCharArray(ssBuf, strg.length() +1);
    strg = String(pwList[modemNo]);
    strg.toCharArray(pwBuf, strg.length() +1);
    trys = 0;
    Serial << "Modem: " << String(ssBuf) << endl;
    WiFi.begin(ssBuf, pwBuf);
    while(trys < 30){ // wait while its connecting
      if(WiFi.status() == WL_CONNECTED) {
        Serial << endl << "Connected to: " << String(ssBuf) << endl;
        // mqtt setup
        client.setServer(serverIP, serverPort);
        while(!client.connected()) {
          client.connect(MyMqttId, userID, userPw);
          waitSecs(3);
        }
        doSubscribe();  // subscribe to topics
        
        if(MyConnection == 'N') MyConnection = 'W';
        EEPROM.write(ModemLoc, modemNo);  // save modem No in EE
        EEPROM.write(EEValid, 'M');
        EEPROM.commit();
        return;
      }
      Serial << "."; delay(500);
      trys ++;
    }
    modemNo ++;   // next modem
    if(modemNo > 4) modemNo = 0;
    Serial << endl;
  }
}
//=================================================
void disconnectWifi() {
  WiFi.persistent(false);
  client.disconnect();    // Mqtt disconnect
  WiFi.disconnect();      // wifi off
  WiFi.mode(WIFI_STA);    // AP off
  ModemFound = false;
  MqttConnected = false;  // signal Mqtt off
  MyConnection = 'N';     // set to reconnect
}
//=================================================
void doConnect() {
  disconnectWifi();
  if(EEPROM.read(EEValid) == 'M') {   // get values from EEPROM
    modemNo = EEPROM.read(ModemLoc);
    ModemFound = true;
  } else {
    modemNo = 0;
    ModemFound = true;
  }
  if(ModemFound) {
    Serial << "ModemNo: " << modemNo << " ModemFound: " << 
    ((ModemFound)?"true":"false") << endl;
    waitSecs(5);
    wifiConnect();
  }
}
//=================================================
void doReConnect() {
  Serial << "Restarting" << endl;
  ESP.eraseConfig();
  //ESP.restart();
  ESP.reset();
}  
//=================================================
void setup() {
  //pinMode(BUILTIN_LED, OUTPUT);     // Initialize the BUILTIN_LED pin as an output
  Serial.begin(115200,SERIAL_8N1,SERIAL_TX_ONLY);
  Serial.flush();
  EEPROM.begin(256);
  delay(100);
  doConnect();
}
//=================================================
void loop() {
  client.loop();
  if(WiFi.status() != WL_CONNECTED) {
    Serial << "Loop: WiFi Lost " << endl;
    doConnect();
  }
  if (!client.connected()) {
    Serial << "Loop: Client Lost Err=" << client.state() << endl;
    if(WiFi.status() == WL_CONNECTED) {
      while(!client.connected()) {
        Serial << "trying client connect" << endl;
        client.connect(MyMqttId, userID, userPw);
        waitSecs(3);
        if(WiFi.status() != WL_CONNECTED) {
          doConnect();
          return;
        }
      }
      doSubscribe();
    } else doConnect();
  }
}
//=================================================
void mqttMessages(char* topic, byte* payload, unsigned int length) {
  String top; String msg = "";
  top = String(topic);
  for(uint8_t i=0; i<length; i++) msg += char(payload[i]);
  Serial << "Mqtt, top: " << top << " msg: " << msg << endl;
}

i've been trying to solve this problem for 3 days solid, the doReconnect() was an attempt to stop the connection logic from cycling through all of the modems by erasing the ESP's config and rebooting the chip instead of disconnecting and reconnecting, that didn't work either

HELPPPPPPPPP i'm loosing hair by the handfull
thanks for any pointers / help
dave

romansavrulin, 2018-03-07:
@Hokahonay try latest git version or rollback to 2.3.0

Hokahonay, 2018-03-08:
@romansavrulin thank you, i am using latest git version, on reflection the problems only started happening when i upgraded from 2.3.0, i'll roll back and try it

Hokahonay, 2018-03-08:
@romansavrulin in the end i had no need to roll back:
For anyone else 'SUFFERING' with this problem my solution turned out to be quite simple
after each client.publish()
and each 'message received'
place client.loop()
with just this my nodes that were showing the problem have been working well for more than 2 hours
if there is any change i will comment back here again
:-)

devyte, 2018-03-08:
Closing as resolved.

closed 2018-03-08 by devyte