esp8266/Arduino #4469: WiFiClientSecure can't connect to shared hosting with SNI

bhaggya55, 2018-03-06:
Configuration:-
- GoDaddy Linux Classic **Shared** Hosting account with primary domain 'example.com'
- UCC SSL certificate installed for 'example.com' with 5 SANs (Subject Alternative Names) for
  - _example1.com, example2.com, example3.com, example4.com, example5.com_
- TLS1.2
- 4 SANs out of above 5 are 'Hosted Domains' in the above mentioned _Shared_ hosting account
- ESP8266 v2.4.0 with ArduinoIDE running example _HTTPSRequestCACert_

If I try to connect to _example.com_ which is the **primary domain** of my hosting account, I see the html response. If I try to connect to _example1.com_, in serial monitor I get:-
```
State:	sending Client Hello (1)
Alert: alert - (unknown 112)
Error: SSL error 112
Alert: unexpected message
Error: SSL error 112
Alert: close notify
connection failed
```
When i checked error 112 out, it means UNRECOGNIZED NAME. Then I found [this](https://stackoverflow.com/questions/7615645/ssl-handshake-alert-unrecognized-name-error-since-upgrade-to-java-1-7-0) Java issue. For me it was caused by GoDaddy shared host not returning a match for the WiFiClientSecure [SNI](https://en.wikipedia.org/wiki/Server_Name_Indication) thus the axTLS handshake fails (even though error 112 is just a warning).

For SNI to work, the Apache server must have a virtual host entry in its configuration. I called their customer support and was told that they can not configure virtual hosts for 'hosted domains' (any domain hosted in the account other than the primary domain _example.com_). In fact my primary domain is just a virtual host in their shared Apache server which makes sense.

The workaround for this is to let axTLS do the SSL handshake without a SNI. To do that one has to comment out the call to axTLS function `ssl_ext_set_host_name(ext, hostName);` inside `WiFiClientSecure.cpp`. Here is the response after that:-
```
State:	sending Client Hello (1)
State:	receiving Server Hello (2)
State:	receiving Certificate (11)
...
// certificate chain prints here...
...
State:	receiving Server Hello Done (14)
State:	sending Client Key Exchange (16)
State:	sending Finished (16)
State:	receiving Finished (16)
Server certificate verified

GET / HTTP/1.1
Host: example1.com
Accept: */*
User-Agent: ESP8266
Connection: Close

request sent

HTTP/1.1 200 OK
Date: Tue, 06 Mar 2018 11:23:27 GMT
Server: Apache
Accept-Ranges: bytes
Content-Length: 62
Connection: close
Content-Type: text/html

<html>
    <head></head>
    <body>Hello There!</body>
</html>
```
I'm sure many of you like me out there have inexpensive shared hosting plans so I thought this information might come handy to you. Until @igrr finds time to include a way to switch off SNI in the call to his `WiFiClientSecure::connect(const String host, uint16_t port)` function, we will have to use this workaround if we need to connect to shared hosts without a virtual host server config.

igrr, 2018-03-06:
Another possible option (which does not need code changes) is to manually resolve the host name to IP, and then use IP address when connecting with WiFiClientSecure.

devyte, 2018-07-13:
Closing due to merged bearssl (axtls will be deprecated), and due to previous comment.

closed 2018-07-13 by devyte

wouterds, 2020-01-31:
Would be nice if we could port this https://github.com/espressif/arduino-esp32/issues/550

d-a-v, 2020-01-31:
Server Name Indication is supported by BearSSL.
Can you open a new issue and describe your request there ?