esp8266/Arduino #4472: digitalRead(MISO) does not read state of HMISO pin

LotusOttawa, 2018-03-06:
I'm porting Arduino SPI code over to the ESP8266 and in the Arduino code there is use of digitalRead(MISO) commands which always read low for the ESP8266 HMISO pin (GPIO12) even though the pin is at a high state.

SPI.transfer(x), digitalWrite(SS,LOW) commands work but digitalRead(MISO) does not.

Here's my SPI header/setup:

#include <SPI.h>
pinMode(SS,OUTPUT);
SPI.beginTransaction(SPISettings(8000000, MSBFIRST, SPI_MODE0));
SPI.begin();

devyte, 2018-03-08:
Closing due to lack of info.
Please open a new issue, and fill in the requested info.

closed 2018-03-08 by devyte

LotusOttawa, 2018-03-08:
I'm not sure what additional information is required?

Setup SPI as shown in my message above. Pull the MISO pin 'high' and execute the digitalread(MISO) command. It comes back with a 'low' result.

Simply pu, digitalread does not work on MISO for esp8266 but DOES work on arduino mini.

Please let me know what additional information you would like and I will provide it.