esp8266/Arduino #4473: Wrong certificate retrieved by WifiClientSecure

kjav, 2018-03-06:
### Basic Info

Hi, I have been trying to make an https request using ESP8266WiFi, on a subdomain of my website.

#### Hardware
Hardware:		ESP-12
Core Version:      	unknown

### Description

_Note: Real domains replaced with example domains._

I have created a subdomain, `sub.domain.com`, of my root website `domain.com`. The **root website** has a security certificate with sha1 hash `17 59 29 DF BF 6D 2D 7F 8D 3B 58 D8 45 87 32 C8 05 3D E6 E5`, which is generated using letsencrypt,
and the **subdomain** has a security certificate with a sha1 hash `88 E6 9D D0 7D A4 D0 E2 8B B5 11 A0 60 1F 4F 02 83 47 9E 51` which was generated using OpenSSL to have a 100 year lifespan. These certificate fingerprints were obtained from inspecting the certificates in the browser.

When I run my code (below), the logs show that it is testing the sha1 hash of the wrong certificate - the root certificate, and not the subdomain certificate. Why is this?

Thanks for your help!

### Settings in IDE

Module:  			ESPino
Flash Size: 		4MB/1MB
CPU Frequency:	80Mhz
Flash Mode:		DIO
Flash Frequency:	921600 baud
Reset Method:		ck

### Sketch

```cpp
#define DEBUG_SSL
#define DEBUGV

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>

const char* ssid = "........";
const char* password = "........";

const char* host = "sub.domain.com";
const int httpsPort = 443;

// Use web browser to view and copy
// SHA1 fingerprint of the certificate
const char* fingerprint = "88:E6:9D:D0:7D:A4:D0:E2:8B:B5:11:A0:60:1F:4F:02:83:47:9E:51";

void setup() {
  Serial.begin(115200);

  Serial.setDebugOutput(true);

  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("WiFi connected");

  // Use WiFiClientSecure class to create TLS connection
  WiFiClientSecure client;

  if (!client.connect(host, httpsPort)) {
    Serial.println("connection failed");
    return;
  }

  if (client.verify(fingerprint, host)) {
    Serial.println("certificate matches");
  } else {
    Serial.println("certificate doesn't match");
  }
}

void loop() {
}

```

### Debug Messages

```

WiFi connected
connecting to sub.domain.com
please start sntp first !
cert FP: 17 59 29 DF BF 6D 2D 7F 8D 3B 58 D8 45 87 32 C8 05 3D E6 E5 
test FP: 88 E6 9D D0 7D A4 D0 E2 8B B5 11 A0 60 1F 4F 02 83 47 9E 51 
certificate doesn't match

```

igrr, 2018-03-07:
Please enable debug output (in tools menu, set Debug Level: SSL, Debug Port: Serial, and also add `Serial.setDebugOutput(true);` after Serial.begin), and attach the log you get when connecting. 

Also you say that `Core Version: unknown`. Please check what the version is. If you have installed it using boards manager, go there and check the installed version number. If you have installed it using git, run `git describe` in the core directory.

earlephilhower, 2018-03-09:
@kjav  In the mode you're using, axtls isn't looking at the cert at all, but do be aware that
> ...which was generated using OpenSSL to have a 100 year lifespan...

The certificate end date will be subject to overflow and should you try and validate it (not its sha1 fingerprint) it'll not work.  Certs past 2038 have the UNIX signed 32-bit time overflow problem.

devyte, 2018-07-13:
Closing due to lack of feedback, and merge of bearssl.

closed 2018-07-13 by devyte