esp8266/Arduino #4476: WifiClientSecure cause wifi evt 7 and reassign ip from dhcp

dariocdj, 2018-03-07:
### Basic Infos

#### Hardware
Hardware:	   ESP-12F
Core Version:      2.4.0

### Description

I'm using esp12 from lots of time with Souliss Framework, and after introducing Universal Telegram Bot that use WifiClientSecure. Normally i set static ip address, and after some hours i find ip address assigned from local dhcp ? why ? and find also lots of wifi evt 7 in logs.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/1MB
CPU Frequency:		80Mhz and tried 160
Flash Mode:			qio
Flash Frequency:	40Mhz?
Upload Using:		OTA /
Reset Method:		ck 

Thanks a lot
Dario Cdj

dariocdj, 2018-03-08:
wifi evt: 7
wifi evt: 7
ip:0.0.0.0,mask:255.255.255.0,gw:192.168.0.1
wifi evt: 3
ip:192.168.0.171,mask:255.255.255.0,gw:192.168.0.1
wifi evt: 3

This is whats happen

jp112sdl, 2018-04-06:
Having the same issue!

dariocdj, 2018-04-09:
At the moment i've solved with a workaround : every 90sec i check actual ip and compare with manual ip assigned, if it is different i assign it again....

jp112sdl, 2018-04-10:
Crazy stuff... :/ 

Starting up the esp: 
```
SDK:2.2.1(cfd48f3)/Core:2.4.1-42-g438d3f1/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1)
wifi evt: 2
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 7
cnt
connected with Embedded-VPN, channel 1
dhcp client start...
wifi evt: 0
```

 Here I do my WiFi.begin(); ->
```
state: 5 -> 0 (0)
rm 0
sleep disable
STUB: dhcp_stop
.wifi evt: 1
STA disconnect: 8
.....scandone
state: 0 -> 2 (b0)
.state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 7
cnt

connected with Embedded-VPN, channel 1
ip:192.168.2.11,mask:255.255.255.0,gw:192.168.2.1
ip:192.168.2.11,mask:255.255.255.0,gw:192.168.2.1
wifi evt: 0
wifi evt: 3
```
now it is connected with the static ip 192.168.2.11.

after a few seconds I get
```
ip:192.168.2.71,mask:255.255.255.0,gw:192.168.2.1
wifi evt: 3
pm open,type:0 0
```
and the esp now is connected with a dhcp assigned ip.

d-a-v, 2018-09-04:
This should be fixed in latest release. Do you confirm ?

devyte, 2018-11-12:
Closing due to lack of feedback.

closed 2018-11-12 by devyte