esp8266/Arduino #4479: digitalRead(MISO) does not read state of HMISO pin

LotusOttawa, 2018-03-08:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:				NodeMCU v1.0 (ESP-12E)
- Core Version:			2.4.1
- Development Env:		PlatformIO
- Operating System:		Windows 10

### Settings in IDE

- Module:  			Nodemcu
- Flash Mode:			qio
- Flash Size: 		        4 MB
- lwip Variant:                  v2
- Reset Method:		nodemcu
- Flash Frequency:	        40 MHz
- CPU Frequency:		80 MHz
- Upload Using:		SERIAL
- Upload Speed:              921600

### Problem Description

I'm porting Arduino SPI realated code over to the ESP8266 and in the code there is use of digitalRead(MISO) commands. This command reads the correct static state of the MISO pin on the Arduino Mini but on NodeMCU it always reads LOW even though I pull the ESP8266 HMISO pin (GPIO12) high.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

#include <Arduino.h>
#include <SPI.h>

void setup() {
    pinMode(SS,OUTPUT);
    SPI.beginTransaction(SPISettings(8000000, MSBFIRST, SPI_MODE0));
    SPI.begin();
}

void loop() {
    digitalRead(MISO);
}

earlephilhower, 2018-03-08:
The MISO pin is NOT an input on the ESP8266 in this mode, probably due to HW limitations as the SPI unit is taking the pins over.

See the line:
https://github.com/esp8266/Arduino/blob/7f0b9d1be9bd3b88fe0e2a26645ddde0e6df966a/libraries/SPI/SPI.cpp#L77

devyte, 2018-07-13:
Closing due to previous commment.

closed 2018-07-13 by devyte