esp8266/Arduino #4483: ESP32 MQTT FATAL ERROR QUEUE.H MISSING

kumareshkc, 2018-03-09:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:			[ESP-32 DEV KIT C]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[ESP32 Dev Module]
- Flash Mode:			[dio]
- Flash Size: 		        [4MB]
- lwip Variant:                   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description
I have a **ESP32 DEV KIT C** module. I am trying to establish a mqtt connection with dioty.co but I am constantly getting this error in my arduino ide for compiling esp32.I am using the **PubSubClient library by Nick O'Leary**. i tried to upload an example code but i am unable to compile the code due to the fatal error occured in the IDE "**ESP8266WiFi\src/ESP8266WiFiType.h:26:19: fatal error: queue.h: No such file or directory** ". i reinstalled the board(esp8266) using .json extension in preferences,by using board manager , by git clone. I reinstalled the arduino ide in different locations but the same error occurs. I have checked all the similar issues and **NONE OF THE SOLUTION WORKS IN MY CASE**. please provide me a proper solution. I installed the ESP32 CORE by using the Git Clone method and the board is working fine.

### Sketch

```cpp
/*
 Basic ESP8266 MQTT example

 This sketch demonstrates the capabilities of the pubsub library in combination
 with the ESP8266 board/library.

 It connects to an MQTT server then:
  - publishes "hello world" to the topic "outTopic" every two seconds
  - subscribes to the topic "inTopic", printing out any messages
    it receives. NB - it assumes the received payloads are strings not binary
  - If the first character of the topic "inTopic" is an 1, switch ON the ESP Led,
    else switch it off

 It will reconnect to the server if the connection is lost using a blocking
 reconnect function. See the 'mqtt_reconnect_nonblocking' example for how to
 achieve the same result without blocking the main loop.

 To install the ESP8266 board, (using Arduino 1.6.4+):
  - Add the following 3rd party board manager under "File -> Preferences -> Additional Boards Manager URLs":
       http://arduino.esp8266.com/stable/package_esp8266com_index.json
  - Open the "Tools -> Board -> Board Manager" and click install for the ESP8266"
  - Select your ESP8266 in "Tools -> Board"

*/

#include <ESP8266WiFi.h>
#include <PubSubClient.h>

// Update these with values suitable for your network.

const char* ssid = "........";
const char* password = "........";
const char* mqtt_server = "broker.mqtt-dashboard.com";

WiFiClient espClient;
PubSubClient client(espClient);
long lastMsg = 0;
char msg[50];
int value = 0;

void setup() {
  pinMode(BUILTIN_LED, OUTPUT);     // Initialize the BUILTIN_LED pin as an output
  Serial.begin(115200);
  setup_wifi();
  client.setServer(mqtt_server, 1883);
  client.setCallback(callback);
}

void setup_wifi() {

  delay(10);
  // We start by connecting to a WiFi network
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void callback(char* topic, byte* payload, unsigned int length) {
  Serial.print("Message arrived [");
  Serial.print(topic);
  Serial.print("] ");
  for (int i = 0; i < length; i++) {
    Serial.print((char)payload[i]);
  }
  Serial.println();

  // Switch on the LED if an 1 was received as first character
  if ((char)payload[0] == '1') {
    digitalWrite(BUILTIN_LED, LOW);   // Turn the LED on (Note that LOW is the voltage level
    // but actually the LED is on; this is because
    // it is acive low on the ESP-01)
  } else {
    digitalWrite(BUILTIN_LED, HIGH);  // Turn the LED off by making the voltage HIGH
  }

}

void reconnect() {
  // Loop until we're reconnected
  while (!client.connected()) {
    Serial.print("Attempting MQTT connection...");
    // Attempt to connect
    if (client.connect("ESP8266Client")) {
      Serial.println("connected");
      // Once connected, publish an announcement...
      client.publish("outTopic", "hello world");
      // ... and resubscribe
      client.subscribe("inTopic");
    } else {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.println(" try again in 5 seconds");
      // Wait 5 seconds before retrying
      delay(5000);
    }
  }
}
void loop() {

  if (!client.connected()) {
    reconnect();
  }
  client.loop();

  long now = millis();
  if (now - lastMsg > 2000) {
    lastMsg = now;
    ++value;
    snprintf (msg, 75, "hello world #%ld", value);
    Serial.print("Publish message: ");
    Serial.println(msg);
    client.publish("outTopic", msg);
  }
}
```

### Debug Messages

```

In file included from C:\Users\Kumaresh\Documents\Arduino\libraries\ESP8266WiFi\src/ESP8266WiFi.h:33:0,

                 from C:\Users\Kumaresh\Documents\Arduino\libraries\PubSubClient\examples\mqtt_esp8266\mqtt_esp8266.ino:26:

C:\Users\Kumaresh\Documents\Arduino\libraries\ESP8266WiFi\src/ESP8266WiFiType.h:26:19: fatal error: queue.h: No such file or directory

compilation terminated.

exit status 1
Error compiling for board ESP32 Dev Module.
```

RudyFiero, 2018-03-09:
What has this got to do with the ESP8266? It is not the same as the ESP32. Where does it say that ESP32 is supported in this repository?

devyte, 2018-03-09:
Previous comment is correct: this repo is for the ESP8266, not your ESP32 based module. You're using the wrong core.
Closing as user error.

closed 2018-03-09 by devyte

rtek1000, 2019-03-20:
Hello,

In case someone else gets confused trying to compile the code from ESP8266 to ESP32:

See this nice code example (link below) for ESP8266, which can be compiled for ESP32 with minor changes:

From (ESP8266):
#include <ESP8266WiFi.h>
To (ESP32):
#include <WiFi.h>

From (ESP8266):
#include <ESP8266WebServer.h>
To (ESP32):
#include <WebServer.h>

From (ESP8266):
ESP8266WebServer server(80); //Server on port 80
To (ESP32):
WebServer server(80); //Server on port 80

Project: "ESP8266 data logging with real time graphs" (AJAX)
https://circuits4you.com/2019/01/11/esp8266-data-logging-with-real-time-graphs/