esp8266/Arduino #4487: Temporary memory drain stalls wifi client

s0170071, 2018-03-09:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
--> no, latest git crashes before I get to this issue. 
I used https://github.com/esp8266/Arduino/commit/85be0f300a0b1e974facdeac697a33268303cde5 from 3 weeks ago

- [x ] I have searched the issue tracker for a similar issue.
- [ x] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			latest 
- Development Env:		[Arduino IDE]
- Operating System:		[Ubuntu]

### Settings in IDE

- Module:  			[Wemos D1 mini r2]
- Flash Mode:			[dio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

- wifi client loses memory at first but works, 
- when memory is very low it stalls, memory returns partly but cannot receive any more. 
- then more memory loss until 900 bytes free. it stays there

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
 
[... 673 lines cut ...]
896
896
896
896
1064
1064
1064
1064
1064
timeout while receiving or too many bytes. 0
1064
1064
connect 192.168.1.41 OK, sending: 0 1 0 0 0 6 126 4 157 7 0 1 
840
840
840
840
840
840
840
840
840
840

```

debsahu, 2018-03-10:
Try this, client.setTimeout(time) before client.connect(). time is in milliseconds. Seems to solve my exception 29 issues.

igrr, 2018-03-14:
Closing, will track in #4497. Please follow the link to that issue and click "subscribe" to receive notifications.

closed 2018-03-14 by igrr