esp8266/Arduino #4491: FunctionalInterrupt latency problem

philbowles, 2018-03-10:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [ X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ X] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [- ] If there is a stack dump, I have decoded it.
- [ X] I have filled out all fields below.

#### Platform

- Hardware:				Wemos D1 mini
- Core Version:			2.4.1
- Development Env:		1.8.5
- Operating System:		Windows10

### Settings in IDE

- Module:  			Wemos D1 mini r2
- Flash Mode:			N/A
- Flash Size: 		        4MB/1MB
- lwip Variant:                  v2 Lower Memory
- Reset Method:		N/A
- Flash Frequency:	        N/A
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:              921600

### Problem Description

I think the signature of attachInterrupt in FunctionalInterrupt.h needs to change, either to
std::function<void(uint8_t)> [ where the uint8_t is the pin state that caused the transition] or better still: std::function<void(uint8_t,uint32_t)> where the 32-bit value is the micros() value closest in time to the initiating event.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
#include<FunctionalInterrupt.h>
#include<functional>
using namespace std;
              
#define PUSHBUTTON 0
uint32_t volatile last=0;

function<void(void)> fvv=[](){
   uint8_t s=digitalRead(PUSHBUTTON);
   uint32_t m=micros();
   uint32_t delta=m-last;
   Serial.printf("T=%d S=%d delta=%d\n",m,s,delta);
   last=m;
};
          
void setup(){
    Serial.begin(74880);
    Serial.printf("Functional Interrupt CHANGE Example, pin=%d\n",PUSHBUTTON); 
    pinMode(PUSHBUTTON,INPUT_PULLUP);
    attachInterrupt(PUSHBUTTON,fvv,CHANGE);
}

void loop(){
}

```

There are two reasons, the first of style/convenience, the second – more serious – regarding problems with latency and missed interrupts.

1)	Style: function<void(void)> - let’s call it “fvv” - has to be explicitly declared as such. If one uses auto fvv=[](){…} or even an anonymous lambda, the compiler objects to the ambigyuity between that and the void function pointer that the “normal” i.e. non-functional attachInterrupt expects.

I suspect there are performance issues that prevent the functional version totally replacing the non-fn version, especially in the classic case of void function pointers vs std::function objects. Should the user be subjected to wrangling this – shouldn’t there be some internal optimisations that select a faster implementation in the case where the functor “decays” to equivalent a void fn pointer? Either way, a change of signature would prevent the ambiguity at least.

2)	Now the real problems: missed interrupts. I appreciate that any interrupt is going to be “missed” if the ISR takes longer than the time between “now” and the next transition. It is incumbent upon us then to “get out soon as possible”. Unless we have a GHz core, we are likely to miss “common” sub-millisecond events like some switch bounces…My thoughts here are on how to minimise and/or remedy them.

All of my comments/observations are based on the following assumptions:
* Where mode = FALLING, pin state @ interrupt time should always == 0
* Where mode = RISING, pin state @ interrupt time should always == 1
* Where mode = CHANGE, pin state @ interrupt time should always alternate between 1 and 0

This will only be true of course if the state is captured at the same time as the cause. Let’s call Ti the time of the interrupt and Tp the time of the pin state read. Any delay between Ti and Tp allows the pin to revert to its previous state. The longer this is the more likely an “incorrect” cause will be read at Tp. The examples show that “when fvv is invoked” is already too late. Thus it must be captured as close to Ti as possible and passed to fvv as a parameter.

I’m no ESP8266 asm expert but I have written an AVR interrupt library and I know I can get the causing pin state with a few machine cycles of the event. Also, users wishing to analyse / decode waveforms need the most accurate transition time…the less code between Ti and Tfvv the better…again the closest micros() to Ti needs to be captured and sent to fvv as a parameter.

So, the proof that Tfvv is too late:
### Debug Messages
```
With attachInterrupt(PUSHBUTTON,fvv,FALLING);
…
T=68373927 S=0 delta=397245
T=68467394 S=0 delta=93467
T=68467606 S=0 delta=212
T=68467706 S=0 delta=100
T=68467805 S=1 delta=99		<-
T=68467982 S=1 delta=177		<-
T=68468245 S=0 delta=263
T=68470630 S=0 delta=2385
T=68757482 S=0 delta=286852
T=68848709 S=0 delta=912
…

With attachInterrupt(PUSHBUTTON,fvv,RISING); [ straight out of the gate!]
Functional Interrupt RISING Example, pin=0
T=9460081 S=1 delta=9460081
T=9460477 S=0 delta=396		<-
T=9540895 S=1 delta=80418
T=9541165 S=1 delta=270
T=17917718 S=1 delta=8376553
T=19087823 S=1 delta=1170105
T=19644068 S=1 delta=556245
T=20070703 S=1 delta=426635
T=20076957 S=1 delta=6254
…
```
There are of course those who will argue “you always know the pin state when a RISING or FALLING int occurs” – but of course only in an ideal world where your ints occur at a slow enough rate. But when CHANGE occurs? Is it a “late” Tp read or the sign of a missed interrupt?

### Debug Messages
```
Functional Interrupt CHANGE Example, pin=0 [ again straight out of the gate…]
T=12533062 S=0 delta=12533062		[ Bouncy keypress		
T=12533454 S=0 delta=392			<- did I miss 0->1 at T<392?
T=12602193 S=1 delta=68739
T=12602874 S=0 delta=681
T=12606496 S=1 delta=3622
T=21534981 S=0 delta=8928485			[ 2nd bouncy keypress
T=21535318 S=1 delta=337			        [ expected
T=25658921 S=0 delta=4123603			[ expected
T=25659229 S=0 delta=308			<- unexpected…"incorrect " 0->1 ?
T=25809738 S=1 delta=150509
```
The pin is pulled up internally – hence its final “correct” state will always be 1. This tends to suggest that the *T=25659229 S=0 delta=308* event is indeed an incorrectly reported (i.e “late” Tp read) real 0->1 change and thus the pulse train can be successfully re-created *if the uSecond times are available…*

devyte, 2018-08-14:
@hreintke is there any chance you could help with this one?

hreintke, 2018-08-14:
@devyte Sure, will look into this and put results into the comments of this issue.

hreintke, 2018-08-15:
@philbowles @devyte 

Summary : 
- Latency is caused by printf in the interrupt routine.
- There is an alternative for explicit declaration.
- I'll do some additional tests for difference between static and functional call. 

BTW : I did implement the "pinvalue and micros()" within the ScheduledInterrupt. With the same arguments as @philbowles. But that is because there is uncertaincy on when the scheduled function will be executed as opposed to instant execution of ISR.

**explicit declaration**

It is correct that the compiler complains about ambiguity when using lambda's. 
Workaround for explicit declaration is the use of std::bind() in the attachInterrupt call.

`attachInterrupt(PUSHBUTTON, std::bind([]() { my ISR code }), CHANGE) `

I will do some additional tests to determine the actual latency difference between static and functional ISR.
After that can be decided whether the static option can be removed.

**Latency / Missed Interrupts**
The high deltas are not because of FunctionalInterrupt latency but because of the use of `Serial.printf("..")` in the ISR.

I updated the sketch to measure with and without the printf.
The first 50 interrupts, -> printing
The second 50 interrupts -> saving  
```
#include "FunctionalInterruptTiming.h"

#include<FunctionalInterrupt.h>
#include<functional>
using namespace std;

#define PUSHBUTTON 14
uint32_t volatile last=0;
uint16_t volatile cnt = 0;

uint16_t volatile dt[200];
uint32_t volatile pv[200];
uint32_t volatile mv[200];

function<void(void)> fvv=[](){
   uint8_t s=digitalRead(PUSHBUTTON);
   uint32_t m=micros();
   uint32_t delta=m-last;
   if (cnt < 50)
   {
	   Serial.printf("T=%d S=%d delta=%d\r\n",m,s,delta);
   }
   dt[cnt] = delta;
   pv[cnt] = s;
   mv[cnt++] = m;

   if (cnt > 99)
   {
	   detachInterrupt(PUSHBUTTON);
   }
   last=m;
};

void setup(){
    Serial.begin(74880);
    Serial.printf("Functional Interrupt CHANGE Example, pin=%d\r\n",PUSHBUTTON);
    pinMode(PUSHBUTTON,INPUT_PULLUP);
    attachInterrupt(PUSHBUTTON,fvv,CHANGE);
}

void loop(){
	if (cnt > 99)
	{
		Serial.printf("Printing saved results----------------------\r\n");
		for (int i = 50;i<cnt;i++)
		{
			Serial.printf("C=%d T=%d S=%d delta=%d\r\n",i, mv[i],pv[i],dt[i]);
		}
		cnt = 0;
	}
}
```

Output from this

```
Functional Interrupt CHANGE Example, pin=14
T=3139058 S=0 delta=3139058
T=3139345 S=0 delta=287
T=3229019 S=0 delta=89674
T=3229155 S=0 delta=136
T=3229234 S=0 delta=79
T=3229312 S=0 delta=78
T=3229389 S=0 delta=77
T=3231217 S=1 delta=1828
T=3232075 S=1 delta=858
T=5793199 S=1 delta=2561124
T=5793384 S=0 delta=185
T=5909067 S=0 delta=115683
T=5909206 S=0 delta=139
T=5909287 S=0 delta=81
T=5909365 S=0 delta=78
T=5909444 S=0 delta=79
T=5909523 S=0 delta=79
T=5911992 S=1 delta=2469
T=7785918 S=1 delta=1873926
T=7786087 S=1 delta=169
T=7786168 S=0 delta=81
T=7786247 S=0 delta=79
T=7786326 S=1 delta=79
T=7786405 S=0 delta=79
T=7789006 S=0 delta=2601
T=7940101 S=0 delta=151095
T=7940632 S=0 delta=531
T=7940714 S=0 delta=82
T=7940793 S=0 delta=79
T=7940991 S=0 delta=198
T=7941243 S=0 delta=252
T=7943297 S=0 delta=2054
T=7946772 S=0 delta=3475
T=7950319 S=0 delta=3547
T=7953710 S=1 delta=3391
T=8700531 S=0 delta=746821
T=8700698 S=0 delta=167
T=8700779 S=0 delta=81
T=8700859 S=0 delta=80
T=8700938 S=0 delta=79
T=8701093 S=0 delta=155
T=8703617 S=0 delta=2524
T=8707088 S=0 delta=3471
T=8897155 S=0 delta=190067
T=8897316 S=0 delta=161
T=8897397 S=0 delta=81
T=8897477 S=0 delta=80
T=8897600 S=0 delta=123
T=8897881 S=0 delta=281
T=8900369 S=0 delta=2488
Printing saved results----------------------
C=50 T=8903840 S=1 delta=3471
C=51 T=9971605 S=1 delta=19189
C=52 T=9971615 S=0 delta=10
C=53 T=9971622 S=1 delta=7
C=54 T=9971628 S=0 delta=6
C=55 T=9971635 S=1 delta=7
C=56 T=9971642 S=0 delta=7
C=57 T=9971649 S=0 delta=7
C=58 T=9971656 S=0 delta=7
C=59 T=9971663 S=0 delta=7
C=60 T=9971670 S=0 delta=7
C=61 T=9971677 S=0 delta=7
C=62 T=9971684 S=0 delta=7
C=63 T=9971692 S=0 delta=8
C=64 T=9971699 S=0 delta=7
C=65 T=9971729 S=0 delta=30
C=66 T=9971739 S=0 delta=10
C=67 T=9971746 S=0 delta=7
C=68 T=9971753 S=0 delta=7
C=69 T=9971760 S=0 delta=7
C=70 T=9971767 S=0 delta=7
C=71 T=9971774 S=0 delta=7
C=72 T=9971781 S=0 delta=7
C=73 T=9971788 S=0 delta=7
C=74 T=9971798 S=0 delta=10
C=75 T=9971826 S=0 delta=28
C=76 T=9971833 S=0 delta=7
C=77 T=9971840 S=0 delta=7
C=78 T=9971848 S=0 delta=8
C=79 T=9971857 S=0 delta=9
C=80 T=9971871 S=0 delta=14
C=81 T=9971878 S=0 delta=7
C=82 T=9971898 S=0 delta=20
C=83 T=9971905 S=0 delta=7
C=84 T=9971912 S=0 delta=7
C=85 T=9971963 S=0 delta=51
C=86 T=9971989 S=0 delta=26
C=87 T=9971999 S=0 delta=10
C=88 T=9972006 S=0 delta=7
C=89 T=9972013 S=0 delta=7
C=90 T=9972020 S=0 delta=7
C=91 T=9972027 S=0 delta=7
C=92 T=9972041 S=0 delta=14
C=93 T=9972053 S=0 delta=12
C=94 T=9972060 S=0 delta=7
C=95 T=9972067 S=0 delta=7
C=96 T=9972074 S=0 delta=7
C=97 T=9972081 S=0 delta=7
C=98 T=9972088 S=0 delta=7
C=99 T=9972096 S=0 delta=8
```

The "saved-only" results are significant lower.

I checked the high number of s=0, delta 7 occurrences with logic analyzer,
They are cause by the bounce creating a lot of "sub-microsecond-spikes" values 0.125 - 0.6 microsecond.

philbowles, 2018-08-21:
Thanks for looking at this, but I'm still in need of knowingt the pin state @ interrupt time...it would be good to have.

devyte, 2018-11-19:
There is no guaranteed way to get the pin state at interrupt time. There is also no guaranteed way to get the micros value at the time of trigger. This is because there is always a small time interval between the interrupt trigger and the time the ISR actually executes, and that time can vary quite a bit due to ISR masking elsewhere. Also, there could be bouncing within that time, which again makes reading the pin state uncertain, which makes providing that as an arg o the callback meaningless.
If you really need this, I suggest just calling digitalRead() at the start of your ISR.
Given that the latency issue is resolved, I'm closing.

closed 2018-11-19 by devyte