esp8266/Arduino #4492: Receive Only WiFi

someone55555, 2018-03-10:
Hello everyone,

in my current project, I'm connecting the esp8266 to a battery, and I'm trying to save as much power as possible (for a longer duration).
is there's a way to cancel the transmit mode, and only to receive data? I guess that receive only save a lot of power.
is it possible by the protocol?

thanks

Pablo2048, 2018-03-10:
Did you read the text under "Delete below"?

philbowles, 2018-03-10:
Simples. Don' t transmit anything... :) What do you think would be being transmitted without you specifically causing it? Pesky stuff like keeping the link alive, maybe? ;^)

Oh and take care choosing your battery type - remember the chip is 3.3v...what hardware ? what board? etc etc (see Pablo's comments above)

This forums is for bug reports, not advice, but here's my advice: ask somewhere else (esp8266.com ?) with A LOT more detail...

closed 2018-03-10 by devyte