esp8266/Arduino #4495: I cannot send data to a server when using lwIP v2

Deimos1994, 2018-03-10:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:				[ESP-12]
- Core Version:			[2_4_1]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu V1.0 | ESP-12E]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4 Higher Bandwidth | v2 Higher Bandwidth] (see below)
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

I cannot send data to IFTTT when using lwIP v2. With lwIP v1.4 it works fine.
I use core version: 2.4.1, but it doesn't send data with 2.4.0 and lwIP v2 also.
I also have a sketch that uses client.connect(serverThingSpeak, 80) and sends data to ThingSpeak, and it also stops working after choosing lwIP v2.
Is it a bug?

Sketch:

```cpp

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266HTTPClient.h>

// Replace with your network credentials
const char* ssid = "myNetwork";
const char* password = "myPassword";
WiFiClient client;

long int previousMilisStormWarning=0;
String Pressure_mmHg = "760.00";
String PressureHourDifference = "1.00";

void setup() {
  Serial.begin(19200);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

  IPAddress ip(192, 168, 2, 213);
  IPAddress gateway(192, 168, 2, 1);
  IPAddress subnet(255, 255, 255, 0);
  WiFi.config(ip, gateway, subnet);

  while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
  }
}

void loop() {
  
  if (millis() - previousMilisStormWarning >= 60000 ) {
      previousMilisStormWarning = millis();
      HTTPClient http;
      http.begin("http://maker.ifttt.com/trigger/storm_warning/with/key/dxnXXXXXXXXXXXXXXzSI");
      http.addHeader("Content-Type", "application/x-www-form-urlencoded");
      http.POST("value1="+String(Pressure_mmHg) + "&value2="+String(PressureHourDifference));
      Serial.println("Response from server:");
      http.writeToStream(&Serial);
      Serial.println();
      http.end();
    }
}

```

### Debug Messages

```
[HTTP-Client][begin] url: http://maker.ifttt.com/trigger/storm_warning/with/key/dxnXXXXXXXXXXXXXXzSI
[HTTP-Client][begin] host: maker.ifttt.com port: 80 url: /trigger/storm_warning/with/key/dxnXXXXXXXXXXXXXXzSI
[HTTP-Client] failed connect to maker.ifttt.com:80
[HTTP-Client][returnError] error(-1): connection refused
Response from server:
[HTTP-Client][returnError] error(-4): not connected

[HTTP-Client][end] tcp is closed
```

Deimos1994, 2018-03-11:
Today I modified my sketch, and now it works with both lwIP v1.4 Higher Bandwidth and lwIP v2 Higher Bandwidth.
I was using this example to make my modifications [BasicHttpClient.ino](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPClient/examples/BasicHttpClient/BasicHttpClient.ino
)
I am still not sure why the previous sketch only works with lwIP v1.4, maybe I'll try to figure it out later.
Any idea?
```
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <ESP8266HTTPClient.h>
#define USE_SERIAL Serial

ESP8266WiFiMulti WiFiMulti;

long int previousMilisStormWarning=0;
float Pressure_mmHg = 760.00;
float PressureHourDifference = 1.00;

void setup() {

  USE_SERIAL.begin(19200);
  // USE_SERIAL.setDebugOutput(true);

  USE_SERIAL.println();
  USE_SERIAL.println();
  USE_SERIAL.println();

  for (uint8_t t = 4; t > 0; t--) {
    USE_SERIAL.printf("[SETUP] WAIT %d...\n", t);
    USE_SERIAL.flush();
    delay(1000);
  }

  WiFi.mode(WIFI_STA);
  WiFiMulti.addAP("myNetwork", "myPassword");

  IPAddress ip(192, 168, 2, 213);
  IPAddress gateway(192, 168, 2, 1);
  IPAddress subnet(255, 255, 255, 0);
  WiFi.config(ip, gateway, subnet);

}

void loop() {

if (millis() - previousMilisStormWarning >= 60000 ) {
      previousMilisStormWarning = millis();
      HTTPClient http;
      http.begin("http://maker.ifttt.com/trigger/storm_warning/with/key/dxnXXXXXXXXXXXXXXzSI");
      http.addHeader("Content-Type", "application/x-www-form-urlencoded");
      http.POST("value1="+String(Pressure_mmHg) + "&value2="+String(PressureHourDifference));
      Serial.println("Response from server:");
      http.writeToStream(&Serial);
      Serial.println();
      http.end();
    }
    
}

```

Deimos1994, 2018-03-11:
**I found a fix for my issue.**
If I add delay(4000); between WiFi.mode(WIFI_STA); and WiFi.begin(ssid, password); it starts to work with lwIP v2.

  
```cpp
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266HTTPClient.h>

// Replace with your network credentials
const char* ssid = "myNetwork";
const char* password = "myPassword";
WiFiClient client;

long int previousMilisStormWarning=0;
float Pressure_mmHg = 760.00;
float PressureHourDifference = 1.00;

void setup() {
  Serial.begin(19200);
  WiFi.mode(WIFI_STA);

 // important delay, it doesn't send data to the server without (only needed if using lwIP v2)
 delay(4000); 

  WiFi.begin(ssid, password);

  IPAddress ip(192, 168, 2, 213);
  IPAddress gateway(192, 168, 2, 1);
  IPAddress subnet(255, 255, 255, 0);
  WiFi.config(ip, gateway, subnet);

  while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
  }
}

void loop() {
  
  if (millis() - previousMilisStormWarning >= 60000 ) {
      previousMilisStormWarning = millis();
      HTTPClient http;
      http.begin("http://maker.ifttt.com/trigger/storm_warning/with/key/dxnXXXXXXXXXXXXXXzSI");
      http.addHeader("Content-Type", "application/x-www-form-urlencoded");
      http.POST("value1="+String(Pressure_mmHg) + "&value2="+String(PressureHourDifference));
      Serial.println("Response from server:");
      http.writeToStream(&Serial);
      Serial.println();
      http.end();
    }
}
```

5chufti, 2018-03-12:
probably the problem is
(WiFi.status() != WL_CONNECTED)
not working correctly in new lwip (several reports), so you try to open connection before wifi established.
alternatively try looping until http.begin() returns true.
in any case: beware of infinite loop stalling your system:
```
  rept = 20;
  while (!client.connect(SERVER, 80) && rept>0) {
    delay(50);
    --rept;
  }
```

Deimos1994, 2018-03-13:
@5chufti I am not an expert, but the sketch is waiting 60 seconds before executing http.begin() is it not enough? Also, my delay(4000) is before WiFi.begin(ssid, password) so it shouldn't have any effect at all, but it has.  Anyways my programming knowledge is limited, but I still curious what is going on.

d-a-v, 2018-03-13:
I will get to this lwip2 issue as soon as I have some spare time.

5chufti, 2018-03-13:
ok, didn't realize the "wait" at begin of loop(). It will not be 60s for the first loop but fair enough.
But still, it should only affect the first connect, for the next loop wifi should allready be online.

As I do almost the same and don't have a problem see my code for connecting and transfer

```
// connecting to a WiFi network
  WiFi.persistent(true);
  WiFi.setSleepMode(WIFI_MODEM_SLEEP);
  WiFi.mode(WIFI_STA);
  WiFi.config(ip, gw, nm, gw, ns);
  WiFi.begin(SSID, WPWD);
  rept = 200;
  while (WiFi.status() != WL_CONNECTED) {
    delay(100);
    if (--rept == 0) return false;
  }
```

```
// transfer
  rept = 20;
  while (!client.connect(SERVER, 80) && rept>0) {
    delay(50);
    --rept;
  }
  if (client.connected()) {
    String postStr = "POST /update HTTP/1.1\n";
    postStr += "Host: " + String(SERVER) + "\n";
    postStr += "Connection: close\n";
    postStr += "X-THINGSPEAKAPIKEY: " + String(APIKEY) + "\n";
    postStr += "Content-Type: application/x-www-form-urlencoded\n";
    postStr += "Content-Length: " + String(dataStr.length()) + "\n";
    postStr += "\n";
    postStr += dataStr;
    //
    client.print(postStr);
    client.flush();
    client.stop();
  }

```
I use the result of client.print() and the server response in debug version 
```
    while (client.connected()) {
      Serial.print(client.readStringUntil('\n'));
```
instead of client.flush(), but as I run the sensor with deep sleep I use an optimistic shortcut for real.

Deimos1994, 2018-03-13:
@5chufti I also use almost the same code to send my data to ThingSpeak in my big sketch, and it didn't work with lwip v2 (without that delay), but I only check if(client.connect(SERVER, 80)) returns true, and then I send the data. The interesting thing that web server and WebSocketsServer work fine, only sending data to IFTTT or ThingSpeak stops working after choosing lwip v2. But that delay(4000) somehow fixed my problem. It would be interesting to see if someone can reproduce the issue using those examples above.

d-a-v, 2018-03-13:
The problem appears that the sketch does not set any dns server address. Then the maker.iffft.com server cannot be resolved, and the "connection refused" (which is wrong, should be something like "unknown host").

With the extra delay it works because by default a dhcp server is requested, and the answer comes up with a dns server in the reply. Then after assigning static addresses, with a luckily valid dns server, the address can be resolved.

Without the delay, this works: ```WiFi.config(ip, gateway, subnet, IPAddress(8, 8, 8, 8));```.

Now we need to check
* what dns is used when lwip-v1.4 is selected.
* why is HTTPClient showing a wrong error message

5chufti, 2018-03-13:
🤦‍♂️  doh ... should have seen that ...

Deimos1994, 2018-03-15:
It is good that we know what causes the issue, but I still don't understand how... I mean ESP8266 will try to connect to AP (the router) after `WiFi.begin(ssid, password);` executed, right? but `delay(4000);` is before `WiFi.begin(ssid, password);` not after it, why it has any effect at all?
After what line of code "a default DHCP server is requested"? if after `WiFi.begin(ssid, password);` then it doesn't make sense to me. Can someone explain it to me?

d-a-v, 2018-03-15:
@Deimos1994 
ESP tries to connect **before** WiFi.config() because (and when) its state is saved in flash: For a faster boot process, the ESP remembers its previous connection credentials (ssid, password) and automatically initiate a dhcp request (right after WiFi.mode(), before begin()).
The (hidden) dhcp request happens to succeed - thanks to your added delay - before you ask for a static configuration.
Your static configuration comes without a dns server address, but the dns provided by dhcp is already here, so your sketch can resolve the iftttt.com address and connect to the matching IP address.

Without the delay, the dns information does not come through dhcp, nor through your static IP request. Without dns, esp cannot resolve ifttt.com to its IP.

You can see that in action with debug mode enabled in the Tool menu of the arduino IDE.

earlephilhower, 2019-01-23:
@d-a-v, is this bug all done?  Looks like it, but you're the expert here.

d-a-v, 2020-05-14:
Closing as sorted out

closed 2020-05-14 by d-a-v