esp8266/Arduino #4496: Support I2S audio input

renne, 2018-03-10:
Please support I2S audio input on the ESP8266 for bidirectional audio streaming. This can be used e.g. for intercoms, VoIP, voice control with audio notifications, ...

devyte, 2018-03-12:
CC @earlephilhower

earlephilhower, 2018-03-12:
I've been looking at some existing code base that enables [I2S input](https://github.com/joextodd/listener).  Seems like it could be added to core_esp8266_i2s without too much grief as it's only setting up another set of dma buffers, starting the slc dma engine, and adding an api to get received data.

Setting this up for the n+2 release for now.

joextodd, 2018-03-12:
If I can help at all with what I have learned from my [listener](https://github.com/joextodd/listener) project, I would be happy to contribute.

earlephilhower, 2018-03-21:
If either of you already has a microphone hooked up, can you look at the new PR #4539 ?  I've taken the code in @joextodd's repo as a start, and combined it with the existing I2S code that support Xmit in Arduino core.  It's based on the same idea as the current xmit, where your app needs to i2s_read_sample() in its loop() to get the data and do work (i.e. requantize to 8-bits, packetize, and send as RTSP or whatever).

It's definitely not done, but it is returning samples that make sense to some degree in my limited testing (resistor from D6->Gnd/VCC on the D1 Mini I have running).  Overflow (i.e. not reading fast enough) will cause some interesting artifacts as-is, for example.

If the ESP8266 has a good HW reference that discusses its I2S block in more detail than the ESP8266-Tech-Ref, I sure as heck can't find it.

earlephilhower, 2018-03-21:
My test app:
````
void setup() {
  Serial.begin(115200):
  i2s_rxtx_begin(true, false);
  i2s_set_rate(10); // Your app should use 44100 or whatever, this lets me debug DMA descriptors
  while(1) {
    uint32_t l, r;
    i2s_read_sample(&l, &r, true);
    Serial.printf("%08x %08x\n", l, r);
  }
}
void loop() { }
````

pashinen, 2018-06-05:
Is there any other mic than SPH0645 as i need frequency range between 20HZ to 20 KHZ.

earlephilhower, 2018-06-05:
@pashinen Sorry, I've only used existing pcbs.  Dual Invensense mikes are in the Google AIY project so they definitely must be low cost.

Closing since the I2S input was added a while back and no feedback from the original requesters.

closed 2018-06-05 by earlephilhower