esp8266/Arduino #4498: NeoPixel show() crashes sometimes, depending on the esp8266 version

BigAl66, 2018-03-10:
I have big trouble using the neopixel library on the esp8266. Some information about my system:

    Hardware: WeMos D1 Mini
    Core Version: 2.4.0, 2.4.0-rc1, 2.4.0-rc2, 2.4.1
    Development Env: Visual Micro Arduino for Visual Studio and Arduino IDE
    Operating System: Windows
    external Hardware: WS2812B data at D5 through a SN74AHCT125N level shifter (135 LEDs)
    used libraries: <WebSocketsServer.h>, <ESP8266WiFi.h>, <ESP8266mDNS.h>, 
                          <DNSServer.h>, <ESP8266WebServer.h>, <WiFiManager.h>, 
                          <WiFiUdp.h>, <ArduinoOTA.h>,  <user_interface.h>, <FS.h>, 
                          <EEPROM.h>, <TimeLib.h>, <Timezone.h>, <Adafruit_NeoPixel.h>

The system is permanent connected to a local WiFi-Network. A WebServer is running and WebSockets are used. Data through the WiFi-Network is only transmitted when it is requested from extern. The refresh rate of the Pixels (show()) is 25 times per second.

After a view minutes or after several hours the system suddenly performs a hardware watchdog reset. I located the position by generating some internal debug information that is stored in the rtc memory. I cleaned up and optimized neopixel's show() commad but - depending on the data timing - the interrupts are blocked their for about 3 ms for each call. I tested several core versions and the interesting thing is, that core version 2.4.0-rc2 is the most stable core for this problem. I had the system running for almost two days before the watchdog reset occures again. With 2.4.1 I gather 10 stable minutes when I have luck. Also the orther versions are much less stable than 2.4.0-rc2.

I wonder, why not more people who use the neopixel library have the same problem.

Some Code:

// main loop
void loop()
{
  // Update over the air handler
  ArduinoOTA.handle();
  // display update
  loopDisplayUpdateCheck();
  // web server
  server.handleClient();
  // web socket server
  webSocket.loop();
}

// check for refresh 
void loopDisplayUpdateCheck()
{
  static unsigned long millis_last = millis();
  unsigned long millis_act = millis();
  if (millis_act - millis_last >= UPDATE_MS)
  {
    // get the time info
    time_t utcTime = now();                                    // utc 
    time_t localTime = utcTime + stp.timeZone.offset * 60;     // localtime (offset is in minutes - we need seconds!)
    if (stp.daylightSavingSetup) {
      localTime = timeZone.toLocal(localTime);                 // offset summer / winter
    }
    tmElements_t Time;
    breakTime(localTime, Time);
    // update the display
    loopDisplayUpdate(Time);
    // store activation time
    millis_last = millis_act;
  }
}

// display update loop
void loopDisplayUpdate(tmElements_t Time)
{
  // update the 7-segment display
  UpdateSevenSegTime(Time);
  // update the hands 
  UpdateHandsTime(Time);
  // show...
  leds.show();
}

This is the whole main loop. The "UpdateSevenSegTime" and "UpdateHandsTime" just prepare the display before it is shown.

The resets are normaly unexpected and during no other activities (no external broweser is connected etc.).

Have anybody an idea what's going wrong? The fact that the stability is core version dependent suggest a problem with the core. I have no more idea where to search...

Regards
  Alex

Makuna, 2018-03-11:
Even if you can fix the issue, you still will not be happy with the results, as you will get flickering.  Try NeoPixelBus library as a replacement for adafruit.

The original esp8266 support for Adafruit library was based on my code, but that way of doing it hasn't worked for close to two years.  I am the author of NeoPixelBus.

devyte, 2018-03-11:
@Makuna thanks for the explanation.
Closing.

closed 2018-03-11 by devyte