esp8266/Arduino #4500: ESP8266WebServer Response time jumps from 0.03 to 1 second if incoming POST payload is >1024 bytes

CodilX, 2018-03-11:
Hello,

I'm trying to post instructions to set LED color values and I've hit a wall. I'm posting a JSON array from my PC and then using that array to set LEDs accordingly on my Nodemcu. I'm using a listener for incoming data `server.on("/command", commandHandler);`

The response time of me posting, receiving, parsing and iterating the JSON array is usually 0.03 seconds. However if I POST to my Nodemcu more than 1024 bytes the response time is always 1 second regardless if I post 1025 bytes or more.

I could send multiple payloads of <1024 bytes however the huge jump from 0.03s to 1s just because of 1 byte makes me wonder if there is some kind of setting I could use to overcome this problem?

5chufti, 2018-03-12:
did you try different lwip settings?

CodilX, 2018-03-12:
@5chufti no I have not. What settings (and how) would you suggest I try?

5chufti, 2018-03-12:
this is a setting in "Tools" menu, chose one from the first three, I cannot recommend one.
It might be relevant for the devs if the problem allready existed in v1.4 and if it is related to smaller buffer in V2 low memory.

CodilX, 2018-03-12:
I've tried all options and the problem remains - the response time is identical. 0.03 - 0.06 seconds for payloads <1024 and 1 second for everything >1024. 

Also, regardless of lwip settings, if I send a payload of <1024 after sending a payload of >1024, the response time is also usually ~1 second, just as if I were sending a payload of >1024. I need to send multiple payloads of <1024 to get the response time down from 1 second.

```
Sending 256  bytes .. 0.06 s.
Sending 512  bytes .. 0.06 s.
Sending 1024 bytes .. 0.06 s.
Sending 1024 bytes .. 0.05 s.
Sending 1536 bytes .. 1.00 s.  <-- dropped response time after >1024 payload
Sending 256  bytes .. 1.00 s.  <-- lowered payload size to <1024
Sending 256  bytes .. 0.99 s.
Sending 256  bytes .. 1.00 s.
Sending 512  bytes .. 1.00 s.
Sending 1024 bytes .. 0.21 s.  <-- improved response time 
Sending 1024 bytes .. 0.06 s.
Sending 1024 bytes .. 0.06 s.
Sending 1536 bytes .. 1.00 s.  <-- dropped response time after >1024 payload
```

5chufti, 2018-03-13:
I hoped that it is only related to the smaller buffer in V2 low memory variant, but it seems it is either more complex or in an other part (http lib?). So at least devs have some more information ...

d-a-v, 2018-03-13:
Can you try with ```server.client().setNoDelay(true)``` on top of your handler ?

CodilX, 2018-03-13:
@5chufti - you can remove the JSON part completely. At first I figured it might be JSON running out of memory or something, but after removing it completely from the code the issues persists, so I posted here. You can actually run this minimal sketch and get the same results:

```
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
ESP8266WebServer server(80);

void setup() {
  WiFi.begin("ssid","pass");
  while (WiFi.status() != WL_CONNECTED) delay(500);  
  server.on("/command", commandHandler);
  server.begin();
}

void commandHandler() { server.send(200, "text/plain", "Received"); }
void loop() { server.handleClient(); }
```

@d-a-v - had no effect :(

d-a-v, 2018-03-13:
> However if I POST to my Nodemcu more than 1024 bytes
Are you posting with your browser ?

In any case would you be able to provide a tcpdump or a wireshark dump of the traffic around this event ?

CodilX, 2018-03-13:
I'm using this PHP code to test it out, as I'll be using something similar in my project down the line. I've also tried not closing the connection between payloads but it didn't have any effect.

```php
$url = 'http://192.168.1.132/command';

$payloads = array(256, 512, 1024, 1536, 256, 256, 256, 512, 1024, 1024, 1024, 1536);
foreach($payloads as $payload) {

	$data = str_pad('1', $payload, '1');
	echo 'Sending '.$payload.' bytes .. ';
	$timeStart = microtime(true); 
	
	$ch = curl_init($url);

	curl_setopt($ch, CURLOPT_POST, 1);
	curl_setopt($ch, CURLOPT_HTTPHEADER, array('Content-Type: text/plain'));
	curl_setopt($ch, CURLOPT_RETURNTRANSFER, true);

	curl_setopt($ch, CURLOPT_POSTFIELDS, $data);
	$do = curl_exec($ch);		

	curl_close($ch);

	echo number_format(microtime(true) - $timeStart, 2). ' s.<br/>';
}
```
I'll try to figure out how to get that dump you want. If someone has a NodeMCU v3 and knows how to do it, just use my minimal sketch a few comments up to replicate the problem.

5chufti, 2018-03-13:
I see the issue in professional hands now, so I'm off ...

d-a-v, 2018-03-13:
Googling 'php curl nagle' gives [that](https://curl.haxx.se/libcurl/c/CURLOPT_TCP_NODELAY.html), stating in doc that Nagle is disabled by default, then below (availability section), enabled by default (nodelay=0 eqv. Nagle enabled, starting from a specific version).

Can you try with Nagle disabled ? with something like ```curl_easy_setopt($ch, CURLOPT_TCP_NODELAY, 1)```, check the example.

If that does not help, a TCP dump is needed, with ```Wireshark``` (or cmdline tool ```tcpdump```) installed on your PC, restricted to your ESP's IP address.

@5chufti you should know your help has always been welcome

CodilX, 2018-03-13:
I've tried taking the client out of the equation multiple times using different methods of sending payloads - PHP, node.js, browser and also shell, all with various settings - same result and unfortunately the `TCP_NODELAY` flag didn't help either.

What flags should I use for `tcpdump` to get the best data you need?

d-a-v, 2018-03-13:
No specific flags. IPs, timestamps, small snap len are usually enough.

Did this issue appeared recently ?

CodilX, 2018-03-13:
I just got this device a week ago to fiddle around with, so I can't tell you when this issue appeared.

Here's the dump

```
18:00:25.391395 IP StarShip.lan.53427 > ESP_1804BE.lan.80: S 223511783:223511783(0) win 64240 <mss 1460,nop,wscale 8,nop,nop,sackOK>
18:00:25.395012 arp who-has StarShip.lan tell ESP_1804BE.lan
18:00:25.395051 arp reply StarShip.lan is-at 74:d4:35:ad:13:23 (oui Unknown)
18:00:25.398437 IP ESP_1804BE.lan.80 > StarShip.lan.53427: S 6510:6510(0) ack 223511784 win 2144 <mss 536>
18:00:25.398513 IP StarShip.lan.53427 > ESP_1804BE.lan.80: . ack 1 win 65392
18:00:25.399366 IP StarShip.lan.53427 > ESP_1804BE.lan.80: P 1:364(363) ack 1 win 65392
18:00:25.408406 IP ESP_1804BE.lan.80 > StarShip.lan.53427: P 1:84(83) ack 364 win 1781
18:00:25.448910 IP StarShip.lan.53427 > ESP_1804BE.lan.80: . ack 84 win 65309
18:00:25.452426 IP ESP_1804BE.lan.80 > StarShip.lan.53427: P 84:92(8) ack 364 win 1781
18:00:25.452502 IP StarShip.lan.53427 > ESP_1804BE.lan.80: F 364:364(0) ack 92 win 65301
18:00:25.452723 IP StarShip.lan.53428 > ESP_1804BE.lan.80: S 2369642941:2369642941(0) win 64240 <mss 1460,nop,wscale 8,nop,nop,sackOK>
18:00:25.455861 IP ESP_1804BE.lan.80 > StarShip.lan.53427: R 92:92(0) ack 365 win 2144
18:00:25.457878 IP ESP_1804BE.lan.80 > StarShip.lan.53428: S 6510:6510(0) ack 2369642942 win 2144 <mss 536>
18:00:25.457926 IP StarShip.lan.53428 > ESP_1804BE.lan.80: . ack 1 win 65392
18:00:25.458509 IP bad-len 0
18:00:25.461693 IP ESP_1804BE.lan.80 > StarShip.lan.53428: . ack 620 win 1525
18:00:25.467199 IP ESP_1804BE.lan.80 > StarShip.lan.53428: . ack 620 win 2061
18:00:25.469264 IP ESP_1804BE.lan.80 > StarShip.lan.53428: P 1:84(83) ack 620 win 2061
18:00:25.508608 IP StarShip.lan.53428 > ESP_1804BE.lan.80: . ack 84 win 65309
18:00:25.510831 IP ESP_1804BE.lan.80 > StarShip.lan.53428: P 84:92(8) ack 620 win 2061
18:00:25.510938 IP StarShip.lan.53428 > ESP_1804BE.lan.80: F 620:620(0) ack 92 win 65301
18:00:25.511177 IP StarShip.lan.53429 > ESP_1804BE.lan.80: S 1620988203:1620988203(0) win 64240 <mss 1460,nop,wscale 8,nop,nop,sackOK>
18:00:25.524120 IP ESP_1804BE.lan.80 > StarShip.lan.53428: R 92:92(0) ack 621 win 2144
18:00:25.524919 IP ESP_1804BE.lan.80 > StarShip.lan.53429: S 6510:6510(0) ack 1620988204 win 2144 <mss 536>
18:00:25.524952 IP StarShip.lan.53429 > ESP_1804BE.lan.80: . ack 1 win 65392
18:00:25.528241 IP bad-len 0
18:00:25.535346 IP ESP_1804BE.lan.80 > StarShip.lan.53429: . ack 537 win 2144
18:00:25.541509 IP ESP_1804BE.lan.80 > StarShip.lan.53429: . ack 1133 win 1548
18:00:25.542681 IP ESP_1804BE.lan.80 > StarShip.lan.53429: . ack 1133 win 2084
18:00:25.543831 IP ESP_1804BE.lan.80 > StarShip.lan.53429: P 1:84(83) ack 1133 win 2084
18:00:25.584259 IP StarShip.lan.53429 > ESP_1804BE.lan.80: . ack 84 win 65309
18:00:25.587223 IP ESP_1804BE.lan.80 > StarShip.lan.53429: P 84:92(8) ack 1133 win 2084
18:00:25.587280 IP StarShip.lan.53429 > ESP_1804BE.lan.80: F 1133:1133(0) ack 92 win 65301
18:00:25.587479 IP StarShip.lan.53430 > ESP_1804BE.lan.80: S 3562910401:3562910401(0) win 64240 <mss 1460,nop,wscale 8,nop,nop,sackOK>
18:00:25.590294 IP ESP_1804BE.lan.80 > StarShip.lan.53429: R 92:92(0) ack 1134 win 2144
18:00:25.590860 IP ESP_1804BE.lan.80 > StarShip.lan.53430: S 6510:6510(0) ack 3562910402 win 2144 <mss 536>
18:00:25.590892 IP StarShip.lan.53430 > ESP_1804BE.lan.80: . ack 1 win 65392
18:00:25.590965 IP StarShip.lan.53430 > ESP_1804BE.lan.80: P 1:131(130) ack 1 win 65392
18:00:25.646459 IP ESP_1804BE.lan.80 > StarShip.lan.53430: . ack 131 win 2014
18:00:26.573713 IP StarShip.lan.53430 > ESP_1804BE.lan.80: F 131:131(0) ack 1 win 65392
18:00:26.574003 IP StarShip.lan.53435 > ESP_1804BE.lan.80: S 2411379980:2411379980(0) win 64240 <mss 1460,nop,wscale 8,nop,nop,sackOK>
18:00:26.576222 IP ESP_1804BE.lan.80 > StarShip.lan.53430: R 1:1(0) ack 132 win 2144
18:00:26.576975 IP ESP_1804BE.lan.80 > StarShip.lan.53435: S 6512:6512(0) ack 2411379981 win 2144 <mss 536>
18:00:26.577022 IP StarShip.lan.53435 > ESP_1804BE.lan.80: . ack 1 win 65392
18:00:26.581698 IP StarShip.lan.53435 > ESP_1804BE.lan.80: P 1:364(363) ack 1 win 65392
18:00:26.670943 IP ESP_1804BE.lan.80 > StarShip.lan.53435: . ack 364 win 1781
18:00:27.574251 IP StarShip.lan.53435 > ESP_1804BE.lan.80: F 364:364(0) ack 1 win 65392
18:00:27.574512 IP StarShip.lan.53436 > ESP_1804BE.lan.80: S 1642034820:1642034820(0) win 64240 <mss 1460,nop,wscale 8,nop,nop,sackOK>
18:00:27.577021 IP ESP_1804BE.lan.80 > StarShip.lan.53435: R 1:1(0) ack 365 win 2144
18:00:27.577937 IP ESP_1804BE.lan.80 > StarShip.lan.53436: S 6516:6516(0) ack 1642034821 win 2144 <mss 536>
18:00:27.577980 IP StarShip.lan.53436 > ESP_1804BE.lan.80: . ack 1 win 65392
18:00:27.581734 IP StarShip.lan.53436 > ESP_1804BE.lan.80: P 1:364(363) ack 1 win 65392
18:00:27.671276 IP ESP_1804BE.lan.80 > StarShip.lan.53436: . ack 364 win 1781
18:00:27.960342 arp who-has ESP_1804BE.lan tell device.lan
18:00:28.573198 IP StarShip.lan.53436 > ESP_1804BE.lan.80: F 364:364(0) ack 1 win 65392
18:00:28.573471 IP StarShip.lan.53437 > ESP_1804BE.lan.80: S 725118112:725118112(0) win 64240 <mss 1460,nop,wscale 8,nop,nop,sackOK>
18:00:28.575949 IP ESP_1804BE.lan.80 > StarShip.lan.53436: R 1:1(0) ack 365 win 2144
18:00:28.576923 IP ESP_1804BE.lan.80 > StarShip.lan.53437: S 6522:6522(0) ack 725118113 win 2144 <mss 536>
18:00:28.576961 IP StarShip.lan.53437 > ESP_1804BE.lan.80: . ack 1 win 65392
18:00:28.581676 IP StarShip.lan.53437 > ESP_1804BE.lan.80: P 1:364(363) ack 1 win 65392
[... cut ...]
18:00:30.942003 IP ESP_1804BE.lan.80 > StarShip.lan.53449: S 6573:6573(0) ack 4079239727 win 2144 <mss 536>
18:00:30.942055 IP StarShip.lan.53449 > ESP_1804BE.lan.80: . ack 1 win 65392
18:00:30.945697 IP StarShip.lan.53449 > ESP_1804BE.lan.80: P 1:131(130) ack 1 win 65392
18:00:31.195601 IP ESP_1804BE.lan.80 > StarShip.lan.53449: . ack 131 win 2014
18:00:31.936695 IP StarShip.lan.53449 > ESP_1804BE.lan.80: F 131:131(0) ack 1 win 65392
18:00:31.939116 IP ESP_1804BE.lan.80 > StarShip.lan.53449: R 1:1(0) ack 132 win 2144
```

d-a-v, 2018-03-15:
That's a pretty big capture to analyse, not sure how many connections are traced, *neither* how many connections occur *inside* one server loop.
I took time to separate them based on peer src port, and check them one by one.
Annotated inside the capture log is what I could see.
(fluent or not, size of the transfer when I could guess it, and big delays when they occur)
There are also capture errors (```IP bad-len 0```)

It happens that when there is a ~1s delay, it is **always** in a session where ESP sends **no data**, still beeing properly replying all ACKs *in time*, and starship taking ~1s to answer its final FIN packet.

Question here: why does the ESP sometimes answer something, sometimes not ? 
This delay is highly correlated with ESP staying silent, not with request size.
Is it on pupose the ESP is silent?

For better understanding, it would be nice to have the ESP doing the dump, HTTP* debug enabled, verbose sketch, so we can have a more precise view of which delay is a painful, and why.
As of now, the ESP can't dump (or at a high price with lwIP expensive and very verbose debug mode enabled), but that's something I planned to do.

At this point there's little more I can say, I can't see where the ESP is the cause if it is, I'm not saying it is not.

```
53427 fluent (256, esp speaks)
18:00:25.391395 IP StarShip.lan.53427 > ESP_1804BE.lan.80: S 223511783:223511783(0) win 64240 <mss 1460,nop,wscale 8,nop,nop,sackOK>
18:00:25.398437 IP ESP_1804BE.lan.80 > StarShip.lan.53427: S 6510:6510(0) ack 223511784 win 2144 <mss 536>
18:00:25.398513 IP StarShip.lan.53427 > ESP_1804BE.lan.80: . ack 1 win 65392
18:00:25.399366 IP StarShip.lan.53427 > ESP_1804BE.lan.80: P 1:364(363) ack 1 win 65392
18:00:25.408406 IP ESP_1804BE.lan.80 > StarShip.lan.53427: P 1:84(83) ack 364 win 1781
18:00:25.448910 IP StarShip.lan.53427 > ESP_1804BE.lan.80: . ack 84 win 65309
18:00:25.452426 IP ESP_1804BE.lan.80 > StarShip.lan.53427: P 84:92(8) ack 364 win 1781
18:00:25.452502 IP StarShip.lan.53427 > ESP_1804BE.lan.80: F 364:364(0) ack 92 win 65301
18:00:25.455861 IP ESP_1804BE.lan.80 > StarShip.lan.53427: R 92:92(0) ack 365 win 2144

53428 fluent (512, esp speaks)
18:00:25.452723 IP StarShip.lan.53428 > ESP_1804BE.lan.80: S 2369642941:2369642941(0) win 64240 <mss 1460,nop,wscale 8,nop,nop,sackOK>
18:00:25.457878 IP ESP_1804BE.lan.80 > StarShip.lan.53428: S 6510:6510(0) ack 2369642942 win 2144 <mss 536>
18:00:25.457926 IP StarShip.lan.53428 > ESP_1804BE.lan.80: . ack 1 win 65392
18:00:25.461693 IP ESP_1804BE.lan.80 > StarShip.lan.53428: . ack 620 win 1525
18:00:25.467199 IP ESP_1804BE.lan.80 > StarShip.lan.53428: . ack 620 win 2061
18:00:25.469264 IP ESP_1804BE.lan.80 > StarShip.lan.53428: P 1:84(83) ack 620 win 2061
18:00:25.508608 IP StarShip.lan.53428 > ESP_1804BE.lan.80: . ack 84 win 65309
18:00:25.510831 IP ESP_1804BE.lan.80 > StarShip.lan.53428: P 84:92(8) ack 620 win 2061
18:00:25.510938 IP StarShip.lan.53428 > ESP_1804BE.lan.80: F 620:620(0) ack 92 win 65301
18:00:25.524120 IP ESP_1804BE.lan.80 > StarShip.lan.53428: R 92:92(0) ack 621 win 2144

53429 fluent (1024, esp speaks)
18:00:25.511177 IP StarShip.lan.53429 > ESP_1804BE.lan.80: S 1620988203:1620988203(0) win 64240 <mss 1460,nop,wscale 8,nop,nop,sackOK>
18:00:25.524919 IP ESP_1804BE.lan.80 > StarShip.lan.53429: S 6510:6510(0) ack 1620988204 win 2144 <mss 536>
18:00:25.524952 IP StarShip.lan.53429 > ESP_1804BE.lan.80: . ack 1 win 65392
18:00:25.535346 IP ESP_1804BE.lan.80 > StarShip.lan.53429: . ack 537 win 2144
18:00:25.541509 IP ESP_1804BE.lan.80 > StarShip.lan.53429: . ack 1133 win 1548
18:00:25.542681 IP ESP_1804BE.lan.80 > StarShip.lan.53429: . ack 1133 win 2084
18:00:25.543831 IP ESP_1804BE.lan.80 > StarShip.lan.53429: P 1:84(83) ack 1133 win 2084
18:00:25.584259 IP StarShip.lan.53429 > ESP_1804BE.lan.80: . ack 84 win 65309
18:00:25.587223 IP ESP_1804BE.lan.80 > StarShip.lan.53429: P 84:92(8) ack 1133 win 2084
18:00:25.587280 IP StarShip.lan.53429 > ESP_1804BE.lan.80: F 1133:1133(0) ack 92 win 65301
18:00:25.590294 IP ESP_1804BE.lan.80 > StarShip.lan.53429: R 92:92(0) ack 1134 win 2144

53430 not fluent (which size? esp does not speak)
18:00:25.587479 IP StarShip.lan.53430 > ESP_1804BE.lan.80: S 3562910401:3562910401(0) win 64240 <mss 1460,nop,wscale 8,nop,nop,sackOK>
18:00:25.590860 IP ESP_1804BE.lan.80 > StarShip.lan.53430: S 6510:6510(0) ack 3562910402 win 2144 <mss 536>
18:00:25.590892 IP StarShip.lan.53430 > ESP_1804BE.lan.80: . ack 1 win 65392
18:00:25.590965 IP StarShip.lan.53430 > ESP_1804BE.lan.80: P 1:131(130) ack 1 win 65392
18:00:25.646459 IP ESP_1804BE.lan.80 > StarShip.lan.53430: . ack 131 win 2014
--- 0.927s --- first starship's distorsion motor brake
18:00:26.573713 IP StarShip.lan.53430 > ESP_1804BE.lan.80: F 131:131(0) ack 1 win 65392
18:00:26.576222 IP ESP_1804BE.lan.80 > StarShip.lan.53430: R 1:1(0) ack 132 win 2144

53435 not fluent (256, esp does not speak)
18:00:26.574003 IP StarShip.lan.53435 > ESP_1804BE.lan.80: S 2411379980:2411379980(0) win 64240 <mss 1460,nop,wscale 8,nop,nop,sackOK>
18:00:26.576975 IP ESP_1804BE.lan.80 > StarShip.lan.53435: S 6512:6512(0) ack 2411379981 win 2144 <mss 536>
18:00:26.577022 IP StarShip.lan.53435 > ESP_1804BE.lan.80: . ack 1 win 65392
18:00:26.581698 IP StarShip.lan.53435 > ESP_1804BE.lan.80: P 1:364(363) ack 1 win 65392
18:00:26.670943 IP ESP_1804BE.lan.80 > StarShip.lan.53435: . ack 364 win 1781
[... 61 lines cut ...]

53448 fluent (1024)
[... cut ...]
18:00:30.942055 IP StarShip.lan.53449 > ESP_1804BE.lan.80: . ack 1 win 65392
18:00:30.945697 IP StarShip.lan.53449 > ESP_1804BE.lan.80: P 1:131(130) ack 1 win 65392
--- 0.250s --- esp is coughing a bit here, takes time to ack a (P)ushed packet
18:00:31.195601 IP ESP_1804BE.lan.80 > StarShip.lan.53449: . ack 131 win 2014
--- 0.741s --- sixth
18:00:31.936695 IP StarShip.lan.53449 > ESP_1804BE.lan.80: F 131:131(0) ack 1 win 65392
18:00:31.939116 IP ESP_1804BE.lan.80 > StarShip.lan.53449: R 1:1(0) ack 132 win 2144
```

aaw20904, 2023-03-18:
Hello! In my own opinion - such a small RAM size (32kBytes) leads to increase time of response.I also tried to write a simple HTTP server - I use examples from the Arduino IDE. The response time of a request was 500mSec. There wasn`t any data process - only 'hello word' in format 'text/plain'. There are a CPU processing all the TCP-IP stack... Data assembling, flow control, wireless Data-Link layer - the CPU is too busy..