esp8266/Arduino #4501: Error Invalid Handshake with SOME websites using HTTPSRequestCACert

jtmoderate876, 2018-03-11:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

Debug level set to everything, SSL TLS_MEM+...

### Problem Description

Cannot connect to some sites with HTTPSRequestCACert.
With debug on I see Invalid Handshake when it tries to Receive Certificate (see debug log below).

I can get it to work with some sites such as api.github.com
But not online.wonderware.com
I've also tried and had trouble with others.

It sounds similar to this log:
https://github.com/esp8266/Arduino/issues/3661
Where that user recommends a change to axtls but I don't have the skills.
 

###  Sketch

```cpp

#include <time.h>
#include <ESP8266WiFi.h>

const char* ssid = "myssid";
const char* password = "mypassword";

const int httpsPort = 443;
const char* host = "online.wonderware.com";
//note: i modified CACert with linux xxd -i DigiCertHighAssuranceEVRootCA.crt.der >cacert.h
// Root certificate used by api.github.com.
// Defined in "CACert" tab.
extern const unsigned char caCert[] PROGMEM;
extern const unsigned int caCertLen;

WiFiClientSecure client;

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.print("connecting to ");
  Serial.println(ssid);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

[... 219 lines cut ...]
:rdi 536, 536
:c0 536, 536
:rn 536
:rd 536, 536, 0
:rdi 536, 536
:c0 536, 536
:rn 255
:rd 255, 255, 0
:rdi 255, 255
:c0 255, 255
State:	receiving Server Hello (2)
State:	receiving Certificate (11)
Error: invalid handshake
:wr 7 7 0
:wrc 7 7 0
Alert: unexpected message
Error: invalid handshake
:wr 7 7 0
:wrc 7 7 0
Alert: close notify
connection failed
connecting to online.wonderware.com

then it repeats
```

jtmoderate876, 2018-03-15:
@igrr Are you too busy to look at this? (trying to HTTPSRequestCACert with sites other than api.github.com such as:
 online.wonderware.com (above)
www.gsmcounters.com (see issue https://github.com/esp8266/Arduino/issues/3661)

jtmoderate876, 2018-03-17:
Anyone having HTTPSRequestCACert  (or similar sketch) working with a different website please list what it works with:

Works with:
api.github.com (verified by jt and others)

Doesn't work with:
online.wonderware.com (verified by jt)

JiriBilek, 2018-03-18:
Hi,
I tested your sketch on iot.bilek.info (verified by Let's Encrypt) and it works. 
I run GET /test.html http request.

Not being an expert in TLS I can't tell it would be useful to you, though...

jtmoderate876, 2018-03-20:
@JiriBilek Thank you for doing that - I'm sure it was a hassle with the DER CACert thing. I appreciate it.
You are connecting and getting past the "Server Hello" stage the we others are failing at with Invalid Handshake - possibly having something to do with axtls.

So HTTPSRequestCACert.ino example works with some websites:
api.github.com (verified by jt and others)
www.bilek.info verified by @JiriBilek

Doesn't work with with others:
online.wonderware.com (verified by jt)
www.gsmcounters.com (see issue #3661)

jtmoderate876, 2018-04-07:
Thankfully, @sukretniy took a version of his fix, changing axtls:
uint8_t *buf = &ssl->bm_data[ssl->dc->bm_proc_index];
replace with:
uint8_t *buf = &ssl->bm_data[ssl->dc->bm_proc_index > 0 ? ssl->dc->bm_proc_index + 6 : ssl->dc->bm_proc_index];

and recompiled it for me because I don't know how and he was kind enough to do so (see his issue #3661) because .
And his fix works for www.wonderware.com too!
It would be good to get this fix into the main code:
igrr/axtls-8266

earlephilhower, 2018-08-04:
Closing this as a duplicate of #3661.  Hopefully @sukretniy can share his source changes so it can go into the esp8266 axtls repo.

closed 2018-08-04 by earlephilhower