esp8266/Arduino #4505: Float not rounded

Gabeki17, 2018-03-12:
- Hardware:			ESP-01
- Core Version:                2.4.0
- Development Env:		[Arduino IDE]
- Operating System:		[Debian & Windows]

### Settings in IDE

- Module:  			[Generic ESP8266]
- Flash Mode:			[DOUT]
- Flash Size: 		[1MB/0MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[OTA & SERIAL]
- Upload Speed:   [230400] (serial upload only)

### Problem Description

Float is not even 0,01 precise.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch
-------------------------------------------------------

void setup() {
  // put your setup code here, to run once:
Serial.begin(115200);
Serial.println("go");
}
float ize = -99999999.99;
void loop() {
  // put your main code here, to run repeatedly:

Serial.println(ize,10);
delay(1000);
}

-----------------------------------------------------------

### Debug Messages

Serial:

go
-100000000.0000000000

liebman, 2018-03-12:
This is not a bug.  You have exceeded the precision of a `float`. It only has 7 decimal digits of precision.

See [this](https://en.wikipedia.org/wiki/Floating-point_arithmetic) for details on how floating point works.

Gabeki17, 2018-03-12:
Hi,

Thx.

But still:

void setup()
{
Serial.begin(115200);
Serial.println("go");
}

float ize = -9.44;

void loop()
{
Serial.println(ize,10);
delay(1000);
}

Serial:

go
-9.4399995803

earlephilhower, 2018-03-12:
Please read @liebman 's link.  This is not a bug with Arduino code but a misunderstanding of how floating point numbers are represented (binary, not decimal digits) and printed (%0.0f to limit output precision).  For more discussion you can go to the forums at www.esp8266.com or elsewhere.  Closing.

closed 2018-03-12 by earlephilhower

liebman, 2018-03-12:
As a demonstration, compiling this on a linux or macos system yields the same conversion:
```c
#include <stdio.h>
int main(int argc, char**argv)
{
    float ize = -9.44;
    printf("%.10f\n", ize);
}
```

```
07:54:17 ~ chris.l$ gcc x.c -o x
07:54:19 ~ chris.l$ ./x
-9.4399995804
```