esp8266/Arduino #4506: Amazon AWS with Custom Domain Name: SSL error 40, SSL handshake failure

wilburgersoftware, 2018-03-12:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

Amazon AWS with Custom Domain Name gets an SSL error 40, SSL handshake failure
Without Custom Domain Name there is no Problem. Also other Domains work well with https.

#### Platform

- Hardware:					ESPWROOM02
- Core Version:			2.4.1
- Development Env:		Arduino IDE|

### Settings in IDE

- Module:  			Nodemcu
- Flash Mode:			dio
- Flash Size: 		4MB
- Flash Frequency:	[40Mhz]
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

Detailed problem description goes here.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>

#define WLAN_SSID    "***"
#define WLAN_PASS    "***"

#define SERVER_HOST  "domaintoaws.com"
#define SERVER_PORT  443

// WiFiFlientSecure for SSL/TLS support
WiFiClientSecure client;

//HTTPS Port fingerprint
const char* fingerprint = "17 59 1b 26 a7 36 7b 95 b0 ef 2d ae b2 58 d4 50 a0 89 d4 bb";

void setup() {
  Serial.begin(115200);
  delay(10);
  Serial.setDebugOutput(true);
 
  // Connect to WiFi access point.
  Serial.print("Connecting to ");
  Serial.println(WLAN_SSID);
  
  delay(1000);
  
  WiFi.begin(WLAN_SSID, WLAN_PASS);
  delay(2000);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("WiFi connected");
  Serial.print("IP address: "); Serial.println(WiFi.localIP());  
}

void loop() {
  
  const char* host = SERVER_HOST;
  const int httpPort = SERVER_PORT;
  
  // create a secure connection using WiFiClientSecure
  WiFiClientSecure client;
  
  if (client.connect(host, httpPort)) {
    
    // verify the signature of the ssl certificate
    if (client.verify(fingerprint, host)) {
      Serial.println("ssl cert matches");
    } else {
      Serial.println("ssl cert mismatch");
    }
  }
  else {
    Serial.println("Connect fail before verification");
  }
  
  delay(60000);
}
```

### Debug Messages

```
[hostByName] request IP for: xxx
[hostByName] Host: xxx IP: 52.11.111.33
:ref 1
State:	sending Client Hello (1)
:wr 96 96 0
:wrc 96 96 0
:sent 96
:rn 7
:rcl
:abort
:rd 5, 7, 0
:rdi 7, 5
:rd 2, 7, 5
:rdi 2, 2
:c0 2, 7
Alert: handshake failure
Error: SSL error 40
Alert: unexpected message
Error: SSL error 40
Alert: close notify
connection failed
:ur 1
```

devyte, 2018-05-29:
@earlephilhower do you have any insight on this one?

earlephilhower, 2018-05-29:
I don't have experience with AWS and custom domain names, but if my reading of the axtls code is right then error code 40 maps to [SSL handshake failure](https://github.com/earlephilhower/axtls-8266/blob/5de79d71e6c66a95f84ade72e0c673aa096e0aa0/ssl/ssl.h#L122).

That is only generated by the [server and client not agreeing on a cipher to exchange info](https://github.com/igrr/axtls-8266/blob/0c3a9f722f11799fbeda1f99f9d9ab77a82a4489/ssl/tls1.c#L1698).  

Maybe using an EC cert with axTLS (not supported) for the custom cert for his domain?  W/o domain where this fails, not really much to do AFAICT.

devyte, 2018-05-30:
@earlephilhower thanks!
The axtls cipher suite won't be upgraded. Given that bearssl is merged, and that axtls is planned for depracation and retirement, I suggest trying the bearssl secure classes. If that still doesn't work, at least there is chance for enhancement with bearssl.
Problems with bearssl should be reported in new issues.
Closing this for now.

closed 2018-05-30 by devyte