esp8266/Arduino #4513: HardwareSerial should be able to report if it has been swapped

eykamp, 2018-03-13:
HardwareSerial has a swap() method, but no way to know if it has been called.  Having an isSwapped() method would be useful, especially as swapped state changes if begin() is called.

Tech-TX, 2021-03-28:
The SDK call is a void (no return to tell that it occurred), so either
  a) track it with a variable in your own code if you can't tell whether you've swapped it already, or
  b) **everyone** has to pay a 4-byte RAM penalty for storing a state that you don't want to track in your code

Seems to make more sense to track what you've done yourself, since the majority of people aren't swapping serial.

[image]

dok-net, 2021-03-31:
Use `bool HardwareSerial::pins(uint8_t tx, uint8_t rx)` for deterministically setting the tx and rx pins instead of `bool HardwareSerial::swap()`, the legal combinations are well-known and documented for `pins()` right in the header file.