esp8266/Arduino #4515: Feature request: Need MDNS Service Name to distinguish between multiple services

chadham, 2018-03-14:
Currently only hostname, IP and port are queryable. Being able to also query service name will allow the ability to manage multiple services.

For example, if multiple mqtt servers are available on the local network, be able to pick the right one based on the service name.

RudyFiero, 2018-03-14:
There is a query on the service name.

https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266mDNS/ESP8266mDNS.h#L91

chadham, 2018-03-14:
I’m referring to the info tag.

For example If query service finds 2 _mqtt.tcp_ services, then the unique info tag can be used to select.

RudyFiero, 2018-03-15:
> I’m referring to the **info tag**.

There is no "info tag" in mdns specification. So how about this since the ESP32 framework already has this and it would be good to have compatible APIs. And if this is not what you are after I do think that this should be an addition to the library.

instance_name: friendly name for your service, like Jhon's ESP32 Web Server. If not defined, default_instance will be used.

http://esp-idf.readthedocs.io/en/latest/api-reference/protocols/mdns.html
```
void add_mdns_services()
{
    //add our services
    mdns_service_add(NULL, "_http", "_tcp", 80, NULL, 0);
    mdns_service_add(NULL, "_arduino", "_tcp", 3232, NULL, 0);
    mdns_service_add(NULL, "_myservice", "_udp", 1234, NULL, 0);

    //NOTE: services must be added before their properties can be set
    //use custom instance for the web server
    **mdns_service_instance_name_set("_http", "_tcp", "Jhon's ESP32 Web Server");**

    mdns_txt_item_t serviceTxtData[3] = {
        {"board","esp32"},
        {"u","user"},
        {"p","password"}
    };
    //set txt data for service (will free and replace current data)
    mdns_service_txt_set("_http", "_tcp", serviceTxtData, 3);

    //change service port
    mdns_service_port_set("_myservice", "_udp", 4321);
}

```

If this is not what you are after then I think you need to be more specific by using proper terms.

chadham, 2018-03-15:
I looked at several mdns entries in the discover browser. Looks like I’m referring to the ability to assign name value pairs to the txt attribute per this 

txt: {var, val} array of strings, used to define properties for your service
Here is a real world example. 

Apple TV._companion-link._tcp.local.

Apple-TV.local:49157
10.0.1.54:49157
fe80::69:a7a3:153b:5869%en0:49157

rpBA = C8:69:CD:6C:40:A9
rpFl = 0x2
rpHA = 5b1ad28c9192
rpHI = b325b1163cc5
rpHN = 8e992310aa82
rpVr = 130.4

devyte, 2018-03-15:
TXT support is already available in the current implementation.
Closing.

closed 2018-03-15 by devyte

RudyFiero, 2018-03-15:
That's right. Have a look at 
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266mDNS/ESP8266mDNS.h#L83

chadham, 2018-03-15:
Thanks. Do I call multiple times to add different kv pairs, or can I pass in a dict structure?

RudyFiero, 2018-03-15:
When I saw that I could set the TXT entry but I could not do a query on it I didn't go any further. You will need to look at the code or just try it.

chadham, 2018-03-15:
Tested adding service text and it works and is exactly what I was looking
for so thanks!

There is a method to query but it is currently private...

MDNSTxt * _getServiceTxt(char *name, char *proto);

Need this to be public unless there is another way to query the service
text that I missed.

GildedHonour, 2020-09-08:
Do these parameters come from the mDNS specification?

```
rpBA = 
rpFl = 
rpHA = 
rpHI = 
rpHN = 
rpVr = 
```

Where are they described in it?