esp8266/Arduino #4517: Documentation: server.send( )/ server.on() / server.arg()

Marcianikus, 2018-03-14:
It is not documented in :
https://arduino-esp8266.readthedocs.io/en/2.4.1/esp8266wifi/readme.html

server.send ()
server.on ()
server.arg ()

I have found it in examples:
https://techtutorialsx.com/2016/10/22/esp8266-webserver-getting-query-parameters/

doubt:
In soft-AP mode, server.send ()
Is it sent to all connected clients?

schweini, 2018-03-30:
> In soft-AP mode, server.send ()
>Is it sent to all connected clients?

While I can't help with the documentation for the server functions, I can confidently say that server.send() only returns data to the currently connected client, whether you are in AP mode or not.

If you have various clients connected at the same time, beware of the problem that their connection will stall while your ESP8266 is busy processing and sending stuff to another client!

rascii, 2019-08-07:
https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WebServer

nghiank97, 2019-11-20:
> [https://github.com/esp8266/Arduino/blob/master/lologists/ESP8266WebServer/README.rst](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/README.rst)

thank you !