esp8266/Arduino #4518: SD Card reader (SPI) and LCD display (I2C) not working together

avdbrink, 2018-03-14:
Combining an SD card reader on SPI and an LCD display on I2C connected to a Wemos D1 R2 with ESP8266 does not seem to be working. 
Both the SD card reader and the LCD display work correctly if used separate, that is: both are connected, but only one is activated in code. But once I combine both of the devices in one sketch, the display stops working after the SD card has been initialised.

Pin connections used is D1 & D2 for I2C and D5, D6, D7 & D8 for SPI.

I've tested the same setup with a regular Arduino Uno and that works. So I assume it has something to do with the ESP8266 library.

Used the most recent 2.4.1 hardware library and even tried some earlier version. 

Minimal code looks like this:
```cpp

// include the SD library:
#include <SPI.h>
#include <SD.h>

// include the wire and LCD I2C libraries:
#include <Wire.h> 
#include <LiquidCrystal_I2C.h>

const int chipSelect = 4;

// set up variables using the SD utility library functions:
Sd2Card card;

// Set the LCD address to 0x27 for a 16 chars and 2 line display
LiquidCrystal_I2C lcd(0x27, 20, 4);

void setup()
{
  // Open serial communications and wait for port to open:
  Serial.begin(115200);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for Leonardo only
  }

  // initialize the LCD
  lcd.begin();
  lcd.clear();
  delay(2000);
  // Turn on the blacklight and print a message.
  lcd.backlight();
  lcd.print("Hello, world!");

  Serial.print("\nInitializing SD card...");

  // we'll use the initialization code from the utility libraries
  // since we're just testing if the card is working!
  if (!card.init(SPI_HALF_SPEED, chipSelect)) {
    Serial.println("initialization failed.");
    return;
  } else {
    Serial.println("Wiring is correct and a card is present.");
  }

  lcd.setCursor(0,3);
  lcd.print("Hello to you too!");
  
  Serial.println("Finished setup()");
}

void loop(void) {

}
```

I see the last Serial.println in the serial monitor, but the display never shows "Hello to you too!", only "Hello, world!".

RudyFiero, 2018-03-15:
If you are using GPIO4 as either SDA or SCL (as is often done) then the problem is this
`const int chipSelect = 4;`

You can not use the same pin for both functions.

avdbrink, 2018-03-15:
Well, thank you very much! That did the trick.
I'm still a little confused, because I do not understand why the SPI interface works correctly while I didn't connect all wires to the pins I told it to use. My chipselect pin is in D8, not in D2. 
Is there an easy explanation for that?

RudyFiero, 2018-03-15:
What happened was the state for the chip select was set so that it was always selected. You had no conflict with other SPI devices since it was the only one.

closed 2018-03-16 by devyte

avdbrink, 2018-03-16:
Okay, that’s clean!
Again: thank you very much!